PX4/PX4-Autopilot #24287: [Bug] Operation timeout, aborting transfer

afwilkin, 2025-02-03:
### Describe the bug

On 1.16 alpha I am seeing an error of "operation timeout, aborting transfer". This is using QGC stable and datalink of sik telemetry 

### To Reproduce

Flash 1.16 on board with sik telemetry radio. and connect over QGC

### Expected behavior

To load parameters and show drone on map

### Screenshot / Media

https://www.loom.com/share/57958b703d93428ba9b223bbeeb9a0fd?sid=d4b89abf-57e4-4e15-9dc8-ac8c554d8074

### Flight Log

I do not have a flight log as this occured before arming. I can change that parameter if need be. 

### Software Version

1.16 alpha2

### Flight controller

holybro pixhawk 4

### Vehicle type

None

### How are the different components wired up (including port information)

GPS to pixhawk4
sik to pixhawk 4
pixhawk4 to pwm esc

Coekjin, 2025-03-20:
I encountered similar issues when using the Pixhawk V3 based flight controller and PX4 v1.15.3 firmware. With the fram and mtd functioning normally, whenever the SD card is removed, errors such as "aborting transfer" and "Rally point transfer fail" appear.

[image]

w2016561536, 2025-12-16:
This bug is caused by DatamanClient readSync and writeSync function (src/lib/dataman_client/DatamanClient.cpp:149 ， release/v1.16). In v1.14, dm_read will check dataman is running or not first. If not, it will return false. But In v1.16,  DatamanClient is separated from dataman, so running status check has been removed. If writeSync and readSync return false. The message disappear.