PX4/PX4-Autopilot #24298: How should the value of the EKF2_EV_CTRL parameter be set in PX4 v1.14?

Tfly6, 2025-02-05:
### Describe the bug

1. I want to use external positioning. In v1.13, I could modify the EKF2_AID_MASK parameter to 24. But in v1.14, the parameter has changed to EKF2_EV_CTRL. I want to modify it, but I found that the default value of EKF2_EV_CTRL is 15. Does that mean external positioning is enabled by default?
2. Furthermore, if I want to use VIO positioning, do these parameters such as EKF2_IMU_CTRL, EKF2_GPS_CTRL, EKF2_OF_CTRL, and EKF2_DRAG_CTRL need to be set?

### Software Version

PX4 : 1.14

### Flight controller

Pixhawk 6c

### Vehicle type

Multicopter

bresch, 2025-02-06:
> the default value of EKF2_EV_CTRL is 15. Does that mean external positioning is enabled by default?

Yes, by default vision fusion starts as soon as there is vision data available.

> do these parameters such as EKF2_IMU_CTRL, EKF2_GPS_CTRL, EKF2_OF_CTRL, and EKF2_DRAG_CTRL need to be set?

EKF2_DRAG_CTRL was disable if you had EKF2_AID_MASK = 24 so you can leave it disabled unless you want to tune the wind estimator.
EKF2_GPS_CTRL is enabled by default. You can disable it as you had it disabled in the aid mask, but if you have no GNSS receiver connected, it won't change anything.
Same for EKF2_OF_CTRL, if you have no optical flow sensor, you don't need to worry about this parameter
For EKF2_IMU_CTRL you can leave it as default unless you want to disable IMU bias estimation for some reason.

Tfly6, 2025-02-06:
@bresch Thank you for your answer!

closed 2025-02-06 by Tfly6