PX4/PX4-Autopilot #24306: [Bug] Unable to select Accel2/Gyro2 as primary IMU

clay350, 2025-02-07:
### Describe the bug

Im not sure if this is a bug or feature request. We are limited to running firmware 1.12.3 on the cube orange and having issues using setting accel2/gyro2 as the primary flight IMU while keeping the other two IMU's active as backups. For us the undampened IMU gives us a much better output. 

The only way we can utilize IMU2 as the primary flight IMU us to disable IMU0/IMU1. We would like to keep IMU0/IMU1 as an active backup.

If I set accel2/gyro2 to maximum priority and the others to "minimum" priority (accel0/gyro0,accel1/gyro1) the machine still defaults to accel0/gyro0 for its primary flight IMU. 

Any help would be appreciated. 

### Vehicle type

None

bresch, 2025-02-14:
Might be a bug on 1.12. One trick would be to change the startup order: https://github.com/PX4/PX4-Autopilot/blob/main/boards/cubepilot/cubeorange/init/rc.board_sensors and start the IMU you want to use first.

> We are limited to running firmware 1.12.3 on the cube orange

Why exactly? Vibration issues on newer versions or something else?

clay350, 2025-02-18:
Im not 100% on the reasoning I have been pushing to upgrade our firmware.  Our mission profile required us to make several modifications to the 1.12.3 code and I think our developers don't want to go through the pain of doing it again on newer firmware. My impression is that the newer firmware handles some items fundamentally differently like the mixer for instance. 

We have always had noisy IMU data that was inconsistent. Eventually we realized that our machines almost always use IMU0 even though the output is markedly worse than IMU2. I think the mechanical internal dampening in the cube is improperly designed and shouldn't be used for larger machines with propellor frequencies below 150hz or so.

closed 2026-01-31 by github-actions[bot]