PX4/PX4-Autopilot #24315: [Bug] Terrain Following Not Working in PX4 Gazebo Simulation Environment

serkanMzlm, 2025-02-11:
### Describe the bug

I conduct tests in the PX4 simulation environment before transitioning to the real system. In this case, I added a distance sensor in the simulation and publish its data at 30 Hz to /fmu/in/distance_sensor via ROS2. I can verify that the vehicle receives this data through QGroundControl (QGC). However, even though the sensor values change in uneven terrain, the drone in the simulation does not follow these changes and continues to fly at a constant altitude

```cpp
using distanceSensorMsg = px4_msgs::msg::DistanceSensor;
...
void DistanceCallback(const laserScanMsg::SharedPtr new_msg)
    {
        RCLCPP_INFO(this->get_logger(), "Distance: %.2f", new_msg->ranges[0]);

        distanceSensorMsg msg;
        msg.timestamp = this->get_clock()->now().nanoseconds() / 1000;
        msg.device_id = 201;
        msg.min_distance = 0.10f;
        msg.max_distance = 220.0f;
        msg.current_distance = new_msg->ranges[0]; // Convert distance to meters
        msg.type = 0;                                             // MAV_DISTANCE_SENSOR_LASER
        msg.h_fov = 0.5 * DEG_TO_RAD;
        msg.v_fov = 0.5 * DEG_TO_RAD;
        msg.orientation = distanceSensorMsg::ROTATION_DOWNWARD_FACING;
        msg.signal_quality = 100;
    
        tf_pub->publish(msg);
    }
```

[image]

Note: Although the sensor data is broadcasting at 30 Hz, it appears as 0.2 Hz in QGC.

[image]

[image]

### To Reproduce

1. Run the Simulation Program 
2. uXRCE-DDS is run
3. Parameters set via QGC:   EKF2_HGT_REF: 2 (range), MPC_ALT_MODE: 1 (Terrain following)
4. "/fmu/in/distance_sensor" sensor data is published
5. Flights are made over terrain with different heights

### Expected behavior

The vehicle should maintain a constant height relative to the ground based on the sensor data.

### Screenshot / Media

- First Point 

[image]

- Second Point 

[image]

Despite the changes in the sensor data, the vehicle did not correct its distance to the ground

### Software Version

PX4: 1.15 version
Simulation: Gazebo Harmonic
ROS2 Humble

### Flight controller

simulation

### Vehicle type

Multicopter

junwoo091400, 2025-02-11:
Could you share the flight log?

serkanMzlm, 2025-02-12:
@junwoo091400 [log](https://review.px4.io/plot_app?log=de9cdc7b-72be-4a9b-a682-260e3906713f)

junwoo091400, 2025-02-12:
> [@junwoo091400](https://github.com/junwoo091400) [log](https://review.px4.io/plot_app?log=de9cdc7b-72be-4a9b-a682-260e3906713f)

When I check the parameter for this log, EKF2_HGT_MODE is set to 1, not 2 as you mentioned. Could you recheck if the parameter was indeed properly set during simulation? I suggest modifying the startup script of the airframe to set the parameters you want to set to desired values as vehicle default.

serkanMzlm, 2025-02-12:
@junwoo091400  I made the corrections you suggested and tried again, but the flight still does not adjust according to the received distance data

junwoo091400, 2025-02-12:
> [@junwoo091400](https://github.com/junwoo091400) I made the corrections you suggested and tried again, but the flight still does not adjust according to the received distance data

Could you share the log for the fixed flight?

serkanMzlm, 2025-02-13:
@junwoo091400  [log](https://review.px4.io/plot_app?log=bf2e49f1-f18d-4646-929d-3915debf1287)

closed 2025-02-24 by serkanMzlm