PX4/PX4-Autopilot #24323: [Bug] [Standard_VTOL] RTL_TYPE 1 does not work in MC

nilsc0, 2025-02-12:
### Describe the bug

Hi there, 

I currently faced an issue using the RTL_TYPE 1 on the standard_vtol in gazebo. When I hovered in MC and switched to Return mode.  When its closer to the VTOL Land Point than the Home Point, the drone just ascended to the RTL_RETURN_ALT and does not continue to follow the VTOL landing pattern. It is then stuck in the air and will only move if another mode, e.g. Land is engaged. 
The SITL in gazebo uses the 1.16.0 (alpha) on the standard_vtol. The same behavior was also reproduced in real-life on a cubeorangeplus with firmware 1.15.1 

### To Reproduce

1. Upload a mission with a Landing Pattern
2. Takeoff in MC and fly closer to the VTOL Land Point than the Home Point
3. Switch mode to Return while still being in MC

### Expected behavior

The drone should follow the landing pattern while staying in MC or at least do anything than just staying in the air. If there is an RC loss with a failsafe action set to RTL, the drone is stuck in the air and will crash when the battery dies. The drone follows the landing pattern when being in Mission mode. The Return mode also works when being in FW while switching to it. The Return mode also works in MC when being closer to the Home Point than to the VTOL Land Point.

### Screenshot / Media

[image]

### Flight Log

https://review.px4.io/plot_app?log=b064999a-28de-4228-bf93-538363b197a9

### Software Version

1.16.0 (alpha) in the SITL 
1.15.1 on the cubeorangeplus IRL

### Flight controller

CubeOrangePlus

### Vehicle type

Hybrid VTOL

### How are the different components wired up (including port information)

Not required due to SITL problem

anaam-wingxpand, 2025-07-14:
I fixed this locally by removing the "Transition to fixed wing if necessary" code from rtl_mission_direct_land.cpp: 

```
else if (_vehicle_status_sub.get().vehicle_type == vehicle_status_s::VEHICLE_TYPE_ROTARY_WING &&
		   _vehicle_status_sub.get().is_vtol &&
		   !_land_detected_sub.get().landed && _work_item_type == WorkItemType::WORK_ITEM_TYPE_DEFAULT) {
		// Transition to fixed wing if necessary.
		set_vtol_transition_item(&_mission_item, vtol_vehicle_status_s::VEHICLE_VTOL_STATE_FW);
		_mission_item.yaw = _navigator->get_local_position()->heading;

		// keep current setpoints (FW position controller generates wp to track during transition)
		pos_sp_triplet->current.type = position_setpoint_s::SETPOINT_TYPE_POSITION;

		new_work_item_type = WorkItemType::WORK_ITEM_TYPE_TRANSITION_AFTER_TAKEOFF;
```

The same bug will also occur for RTL_TYPE 2 if the VTOL is in MC mode, and the same fix will work in the rtl_mission_fast.cpp and rtl_mission_fast_reverse.cpp files.

It's not clear to me why anyone would want "Transition to fixed wing if necessary" during an RTL, so I think this is a safe fix.

mbjd, 2026-09-18:
fixed by https://github.com/PX4/PX4-Autopilot/pull/28399

closed 2026-09-18 by mbjd