PX4/PX4-Autopilot #24324: indoor work in a simulation environment

serkanMzlm, 2025-02-12:
### Describe problem solved by the proposed feature

All PX4 simulations in Gazebo Sim operate with GPS dependency. Additionally, creating an environment with optical flow and distance sensors for GPS-independent scenarios and making the necessary adjustments within this environment could be beneficial. Currently, I am using VIO with an Intel RealSense sensor mounted on the drone, and it works quite well. However, if PX4 used optical flow instead of GPS, we could observe the drone's behavior more effectively in this case

### Describe your preferred solution

With an optical flow and distance sensor under PX4, we can test the drone's GPS-independent flight performance in a simulation environment. Additionally, by integrating external positioning methods such as VIO, we can improve the accuracy and effectiveness of optical flow.

### Describe possible alternatives

More detailed maps and surface features for proper optical flow operation..

[image]

[image]

[image]

[image]

dakejahl, 2025-02-12:
Do you have these worlds running in gz already? It would be great if you could submit a PR to add them to the gz models repo.

I have a WIP branch here for adding an optical flow plugin for gz sim for exactly this reason. The PR is basically done so you're welcome to test it out
https://github.com/PX4/PX4-Autopilot/pull/24153

serkanMzlm, 2025-02-13:
@dakejahl  yes, this world model and drone model are working smoothly in Gazebo. I will submit these works as a PR. I will review your work and test it again. Thank you.

closed 2025-02-19 by serkanMzlm