PX4/PX4-Autopilot #24325: [Bug] It does not work properly in PX4 version 1.15

serkanMzlm, 2025-02-12:
### Describe the bug

I installed a drone with PX4 version 1.15 and added an optical flow and distance sensor for indoor flight. I confirmed that the sensor orientations were correct and performed the necessary calibrations. Then, I tried to arm in position mode, but I kept getting the 'px4 no valid local position estimate' error. Without making any changes, I upgraded to version 1.16, and it started working in position mode.

### To Reproduce

1. I powered up the drone.
2. I checked the sensors data via QGC
3. Drone, I put it in position mode
4. I sent an arm command
5. I got the error "px4 no valid local position estimate'"

### Expected behavior

For PX4 to work properly in position mode, the drone should have been able to arm when optical flow and distance sensors were connected.

### Software Version

PX4: 1.15.0

### Vehicle type

None

serkanMzlm, 2025-02-13:
I needed to enable the EKF2_OF_CTRL value

closed 2025-02-13 by serkanMzlm

bresch, 2025-02-13:
Hmm, maybe we could enable it by default.