PX4/PX4-Autopilot #24334: [Bug] Mag Calibration Without GPS Can Produce Bad Values

AlexKlimaj, 2025-02-13:
### Describe the bug

When calibrating a drone without a GPS fix, the calibration can result in very bad values with high offsets and small scalars.

The default MAG_SPHERE_RADIUS_DEFAULT is 0.2 currently. 
https://github.com/PX4/PX4-Autopilot/blob/b77797b4903e2cb5ba460834f835c94ee76e32ae/src/modules/commander/mag_calibration.cpp#L74

### Screenshot / Media

[image]

### Vehicle type

None

bresch, 2025-02-14:
If the earth mag field goes from 0.25 - 0.65 Gauss, any idea why do we have that value set to 0.2?

AlexKlimaj, 2025-02-14:
I have no clue.

mrpollo, 2025-04-23:
We are going to bump to after the release of v1.16

closed 2026-01-31 by github-actions[bot]