PX4/PX4-Autopilot #24338: [Bug] PX4 cannot maintain altitude in position mode

serkanMzlm, 2025-02-14:
### Describe the bug

I have a quadcopter drone built with PX4. I fly this drone in position mode using an optical flow and a distance sensor. When hovering, there is no problem, and it maintains its position. However, when I move the drone at a speed of 5 m/s, it continuously descends. Starting at an altitude of 5 meters, after moving 20 meters in the X direction, its Z position drops by about 3 meter, and it cannot maintain a stable altitude.  When I stop the vehicle, it slowly climbs back to 5 meters. "In this setup, the altitude sensor is not directly connected to PX4. I read the distance sensor data via Raspberry Pi and publish it as distance_sensor, and I can see this data on QGC. 

### Flight Log

https://review.px4.io/plot_app?log=2bf29ee9-9c62-460e-8b27-317d46bb12db

### Software Version

1.15.4

### Flight controller

Pixhawk 6C Mini

### Vehicle type

None

dakejahl, 2025-02-14:
Similar to https://github.com/PX4/PX4-Autopilot/issues/24058

We are investigating and working on fixing this

TompsonTan, 2025-02-15:
v1.15 and v1.16apha,  I use only baro as height source,  height is not stable in realworld and sitl.

serkanMzlm, 2025-02-16:
@dakejahl There is another point here that I cannot fully understand, and this might be the cause of the issue. I am publishing distance sensor data from a Raspberry Pi. The distance_sensor message contains a variable called "signal_quality," which I populate based on the values specified in the sensor’s datasheet. When I listen to the distance sensor using ros2 topic echo, the signal_quality value fluctuates between 75 and 100. However, when I connect to the PX4 board via QGroundControl, the distance sensor data appears correct, but the signal_quality value is shown as 0. Could this be affecting the behavior? Could it be that PX4 relies on the barometer because it sees signal_quality as 0?

dakejahl, 2025-02-16:
I tested main with gz sim yesterday and it seems that altitude hold is broken with both baro and gps as the primary height reference. Just taking off and hovering for a few minutes I see the altitude fluctuate up to 2m.

serkanMzlm, 2025-02-17:
@dakejahl Yes, it is normal for this to happen when the barometer has priority. However, since my priority is the distance sensor, I don’t understand why this is happening. Secondly, I am wondering why the signal_quality value is shown as 0.

TompsonTan, 2025-02-17:
It seams that EKF2_IMU_CTRL AccelBias has some problem.

only use baro:

EKF2_IMU_CTRL = 7, Altitude not stable.
[image]

EKF2_IMU_CTRL = 5, Altitude is stable.
[image]

bresch, 2025-02-17:
@TompsonTan Could you check the accelerometer values? I remember seeing that in gazebo the accelerometer bias changes a lot in the first seconds after takeoff. @haumarco also experienced this lately.

serkanMzlm, 2025-02-18:
@bresch  I am experiencing this issue on a real vehicle, not in simulation. The vehicle cannot maintain a stable altitude. What could be the reason for this?

omeredemen, 2025-02-18:
@serkanMzlm, @bresch I am experiencing a similar issue. In this part, I am publishing distance sensor data with ROS2, but no matter what signal_quality value I send, it always appears as 0 in the QGC application. What could be the reason for this?

dakejahl, 2025-02-18:
The signal_quality is not getting published in the mavlink message, fixed here https://github.com/PX4/PX4-Autopilot/pull/24357

serkanMzlm, 2025-02-19:
@dakejahl Thank you for your answer

omeredemen, 2025-02-19:
> The signal_quality is not getting published in the mavlink message, fixed here [#24357](https://github.com/PX4/PX4-Autopilot/pull/24357)

yes this will solve the problem thank you

TompsonTan, 2025-02-19:
I switched to commit https://github.com/PX4/PX4-Autopilot/commit/7c7a3c117a9ee71ee1592733352fa852f89bf1cf,   which ekf already use Error-State Kalman Filter,  but before [use global definition of quaternion error],  In read vehicle test, the altitude seams become more stable. Still need more tests.
Maybe the newest ekf still have some issues. @dakejahl @bresch

bresch, 2025-02-19:
The problem is that there are many reasons why the altitude can be incorrect:
- accelerometer bias change/clipping
- baro pressure change depending on flight condition and wind
- GNSS solution shift (e.g.: after gaining SBAS corrections, multipath)
- distance sensor partial obstruction (can easily affect the height if the drone is attempting to hold its height above ground)
- too low accuracy of aiding sources (realistically, the estimate cannot have a centimeter accuracy if the drone is only using an IMU and a barometer with an accuracy of 3m)
- a bug in the code

The list is not exhaustive and there are probably many more reasons why a drone wouldn't hold its altitude. Since @dakejahl was able to trigger such an issue in SITL, let's first investigate this.

dakejahl, 2025-02-22:
I change the SensorGpsSim and SensorBaroSim to match the groundtruth values exactly and I am still seeing the issue. I don't see anything wrong with the accel values. 

[image]

serkanMzlm, 2025-02-25:
@dakejahl  I eliminated the effect of the barometer on altitude by completely disabling it. This way, the system relies solely on the distance sensor and is now able to maintain a stable altitude

bresch, 2025-02-25:
@dakejahl That's probably related to the simulator. For example, I get perfect estimates in SITL jMavSim

[image].

Could you please open a new issue about unreliable height estimates in Gazebo (and provide a replayable log) so we don't mix all issues together?

closed 2025-02-25 by bresch