PX4/PX4-Autopilot #24355: [Bug] GPS failure, notify user

dakejahl, 2025-02-18:
### Describe the bug

We had a customer experience some momentary GPS interference which kicked the vehicle out of position mode and into altitude mode. They reported the issue to us because it was not clear from the log what caused the failure. The log file contains some cryptic warnings.

[image]

After analyzing the log it appears that the speed accuracy exceeded the max threshold. Although I tried to reproduce this in sim and was unable to cause the failure by simply increasing the speed error. It's hard to tell where exactly in ekf code the failure occurs without some kind of indicator. 

### To Reproduce

Experience GPS interference. Maybe a long term goal would be to add a failure injection test mode into sim for simulating GPS interference.

### Expected behavior

There should be an event notification that informs the user at GCS that the vehicle has lost position hold (xy_valid goes false). Ideally it would be even more specific and indicate the exact reason for failure (eg speed accuracy exceeds max threshold). The log file should also contain this notification and we should suppress the other side effect warnings.

### Screenshot / Media

[image]

### Flight Log

https://logs.px4.io/plot_app?log=c25d5ee4-1531-4dc3-9906-637c52488da4

### Software Version

v1.15.0

### Flight controller

ARKV6X

### Vehicle type

Multicopter

jeremyzff, 2025-03-20:
to add it this, Freefly Systems is also having this problem in a few different forms

This log does the same thing when RTK stream on the ground station is stopped. 
 https://logs.px4.io/plot_app?log=9c2fc42d-171b-40fe-ab77-32ac7895d92d

We have had similar problems where small spikes above PDOP threshold also cause this.

Saibernard, 2026-09-23:
@dakejahl I was going through this while working on the selector failover fix and noticed something, and I want to check my understanding before I go further. EKF2 already publishes estimator_gps_status with a flag for each check, so the speed error case from your first log and the PDOP case Freefly mentioned are both sitting in there as a bool. As far as I can tell nothing outside ekf2 reads that message, so it never makes it to a log line or to QGC. Is that right, or is there a consumer I'm missing?

If so, I am happy to pick this up. The idea would be that when local position goes invalid we send one event saying which check failed, log the same line, and suppress the other warnings that show up at the same time so the real reason isn't lost in them. I wouldn't touch what the estimator decides, so whether it should have dropped the velocity in the first place stays a separate question. Let me know if you see it differently.

I can use the GNSS failure injection from #28398 to test it in SITL and the logs here for real cases.

dakejahl, 2026-09-23:
**Claude on behalf of @dakejahl**

@Saibernard Yes, please pick it up.

You're right that nothing subscribes to `estimator_gps_status`, and it's only logged through the logger's `estimator*` catch-all. The same bits are in `estimator_status.gps_check_fail_flags`, which commander does read, but apart from spoofing and jamming it only reports them while disarmed. In flight, a failed quality check shows up only as "GNSS data fusion stopped", with no reason.

The logs here are v1.15, which applied the strict `EKF2_REQ_*` thresholds in flight. Since #24909 (v1.17) the in-flight gate is only fix < 3D, eph/epv > 50 m, sacc > 10 m/s, spoofing and jamming. In the Freefly log, stopping RTK took eph from 0.02 to 3.5 m against `EKF2_REQ_EPH = 1.0`. Local position went invalid at 88.4 s (blind land), and "GNSS data fusion stopped" came at 90.3 s, at Info. Main wouldn't trip there, so what's left is the reporting. Triggering on local position going invalid is the right call: with default timeouts the fusion-stopped event lags it by 2 s.

Please keep it in commander, reading `estimator_status.gps_check_fail_flags`. We've been discussing moving the receiver checks out of EKF2 and into the sensors module. Today a receiver's speed accuracy passes through blending (which takes the minimum across receivers), the EKF buffer, and a check inside every EKF instance before it becomes a bit in `estimator_status`. It would be much simpler to check each receiver once in sensors and leave EKF2 only its estimator-specific checks. That isn't straightforward, as #28520 has shown, and it's a few steps out. When it lands, a commander-side event only has to change its input, and EKF2 will report directly why it isn't fusing GNSS.

Where this is heading (plan: https://github.com/dakejahl/botenook/blob/main/gnss/pipeline_migration.md):

1. Separate GNSS heading from position: #27102
2. Rename gps → gnss (#24399 has the field renames), and give the sensors output its own `VehicleGnss` message
3. Replace blending with selection and failover: https://github.com/PX4/PX4-Autopilot/pull/28798#pullrequestreview-5282016544
4. Move the receiver checks into sensors, with the result carried on each `vehicle_gnss` sample

1 has to land before blending goes, since blending is what keeps CAN moving-base heading working today. The rename goes right after it so everything later is written against the final names, and selection comes before the checks move so we never compute a check result for a blended solution.

Saibernard, 2026-09-27:
Thanks, will do. I'll keep it in commander on estimator_status.gps_check_fail_flags, triggered on local position going invalid, and leave the in-flight gate as it is.

closed 2026-09-29 by dakejahl