PX4/PX4-Autopilot #24370: [Bug] Failed to build SITL

edu-rossrobotics, 2025-02-19:
### Describe the bug

Hi, I'm trying to build a target to run a SITL simulation but I'm getting the following error:

```bash
[14/76] Building CXX object src/modules/simulation/gz_bridge/CMakeFiles/modules__simulation__gz_bridge.dir/GZMixingInterfaceESC.cpp.o
FAILED: src/modules/simulation/gz_bridge/CMakeFiles/modules__simulation__gz_bridge.dir/GZMixingInterfaceESC.cpp.o
/usr/bin/ccache /usr/bin/c++ -DCONFIG_ARCH_BOARD_PX4_SITL -DENABLE_LOCKSTEP_SCHEDULER -DMODULE_NAME=\"gz_bridge\" -DPX4_MAIN=gz_bridge_app_main -D__CUSTOM_FILE_IO__ -D__PX4_LINUX -D__PX4_POSIX -D__STDC_FORMAT_MACROS -Dnoreturn_function="__attribute__((noreturn))" -I/home/edu/siddux-github/PX4-Autopilot/boards/px4/sitl/src -I/home/edu/siddux-github/PX4-Autopilot/platforms/posix/src/px4/common/include -I/home/edu/siddux-github/PX4-Autopilot/build/px4_sitl_default -I/home/edu/siddux-github/PX4-Autopilot/build/px4_sitl_default/src/lib -I/home/edu/siddux-github/PX4-Autopilot/platforms/posix/src/px4/generic/generic/include -I/home/edu/siddux-github/PX4-Autopilot/platforms/common -I/home/edu/siddux-github/PX4-Autopilot/platforms/common/include -I/home/edu/siddux-github/PX4-Autopilot/src -I/home/edu/siddux-github/PX4-Autopilot/src/include -I/home/edu/siddux-github/PX4-Autopilot/src/lib -I/home/edu/siddux-github/PX4-Autopilot/src/lib/matrix -I/home/edu/siddux-github/PX4-Autopilot/src/modules -I/home/edu/siddux-github/PX4-Autopilot/platforms/posix/include -I/home/edu/siddux-github/PX4-Autopilot/build/px4_sitl_default/external/Install/include -isystem /usr/include/gz/transport13 -isystem /home/edu/.local/include -isystem /usr/include/gz/utils2 -isystem /usr/include/gz/msgs10 -isystem /usr/include/gz/math7 -isystem /usr/include/uuid -O2 -g -DNDEBUG -std=gnu++17 -fPIC -g -fdata-sections -ffunction-sections -fomit-frame-pointer -fmerge-all-constants -fno-signed-zeros -fno-trapping-math -freciprocal-math -fno-math-errno -fno-strict-aliasing -fvisibility=hidden -include visibility.h -Wall -Wextra -Werror -Warray-bounds -Wcast-align -Wdisabled-optimization -Wdouble-promotion -Wfatal-errors -Wfloat-equal -Wformat-security -Winit-self -Wlogical-op -Wpointer-arith -Wshadow -Wuninitialized -Wunknown-pragmas -Wunused-variable -Wno-missing-field-initializers -Wno-missing-include-dirs -Wno-unused-parameter -fdiagnostics-color=always -Wno-stringop-truncation -fno-builtin-printf -fno-strength-reduce -Wformat=1 -Wunused-but-set-variable -Wno-format-truncation -fcheck-new -Wreorder -Wno-overloaded-virtual -frtti -fexceptions -O2 -I/usr/include -I/usr/include/uuid -MD -MT src/modules/simulation/gz_bridge/CMakeFiles/modules__simulation__gz_bridge.dir/GZMixingInterfaceESC.cpp.o -MF src/modules/simulation/gz_bridge/CMakeFiles/modules__simulation__gz_bridge.dir/GZMixingInterfaceESC.cpp.o.d -o src/modules/simulation/gz_bridge/CMakeFiles/modules__simulation__gz_bridge.dir/GZMixingInterfaceESC.cpp.o -c /home/edu/siddux-github/PX4-Autopilot/src/modules/simulation/gz_bridge/GZMixingInterfaceESC.cpp
In file included from /usr/include/gz/msgs10/gz/msgs/vector3d.pb.h:12,
                 from /usr/include/gz/msgs10/gz/msgs/convert/Vector3.hh:21,
                 from /usr/include/gz/msgs10/gz/msgs/convert/AxisAlignedBox.hh:20,
                 from /usr/include/gz/msgs10/gz/msgs.hh:23,
                 from /home/edu/siddux-github/PX4-Autopilot/src/modules/simulation/gz_bridge/GZMixingInterfaceESC.hpp:38,
                 from /home/edu/siddux-github/PX4-Autopilot/src/modules/simulation/gz_bridge/GZMixingInterfaceESC.cpp:34:
/usr/include/gz/msgs10/gz/msgs/details/vector3d.pb.h:12:2: error: #error This file was generated by a newer version of protoc which is
   12 | #error This file was generated by a newer version of protoc which is
      |  ^~~~~
compilation terminated due to -Wfatal-errors.
```

Initially I thought it could be related to the version I had installed for the protobuf-compiler package but after removing it and re-run the setup script for ubuntu I end up with the same issue.

### To Reproduce

Using latest main, run the make command:

```bash
make px4_sitl gz_x500
```

### Expected behavior

Build without errors

### Software Version

Latest main branch

### Vehicle type

None

tassalor1, 2025-02-20:
This is known and added here:
https://github.com/PX4/PX4-Autopilot/issues/24369#issue-2863850374

claudiusli, 2025-04-01:
Hi @tassalor1 ,
I'm wondering if this might be a slightly different issue. 
#24369 seems to be part of a larger thread on moving from gazebo classic to gazebo (harmonic)

I think this is something else and I've been encountering a very similar issue.
I'm getting failures on opticalflow but it's also saying there's a mismatch between the protoc version used to create the file and the protobuf version on my system.

There have been similar, older threads https://discuss.px4.io/t/make-px4-sitl-gazebo-error/17982/3 that suggested rolling back the version of protobuf on the system. It seems like the "better" solution would be to regenerate those files with the current version of protoc.

claudiusli, 2025-04-02:
I think [#24651](https://github.com/PX4/PX4-Autopilot/issues/24651) is the actual fix and that's reflected in the correct, but slightly confusing documentation as described in [#3656
](https://github.com/PX4/PX4-user_guide/issues/3656)

tl;dr Until 24651 is done simulation requires Gazebo Classic and, on Ubuntu, that must be version 20.04.

hamishwillee, 2025-04-03:
@claudiusli Not really. #24651 is related to MAVSDK and integrations tests requiring Gazebo-Classic. They simply have never been ported to work with Gazebo. There might be several reasons for this, including things like peripherals that are not supported in the newer simulator.

This is an issue with Gazebo not building cleanly. The issue @tassalor1 points to is likely part of the solution - we need better error reporting if the dependencies haven't been installed at all.

FWIW I ran `make px4_sitl gz_x500`  on WSL2 in Windows 11 on Ubuntu 22.04.5 - no problems at all.