PX4/PX4-Autopilot #24376: [Bug]  Not flashing correct revision firmware to FC.

DroneWuKong, 2025-02-20:
### Describe the bug

When updating the firmware to v1.16.0 Alpha2, after connecting the flight controller and selecting the beta firmware option, everything installs and loads up fine, but when I look at the vehicle summary page it shows that  v1.15 had been flashed. 

Making a custom build of 1.16.0Alpha2, then flashing that firmware locally was my workaround. 

### To Reproduce

1. Open QGC
2. Click the Q icon on upper right > Vehicle Setup > Firmware (Step 1 in picture)
3. Connect your flight controller to the computer
4. When the Firmware setup window pops up, click "Advanced settings" checkbox
5. Select "Beat Testing (beta)" from the drop down (Step 2 in picture)
6. Click "Ok"

### Expected behavior

When the beta firmware is selected, it updates to that revision of the firmware. 

### Screenshot / Media

[image]
[image]
[image]

### Software Version

ver all
HW arch: PX4_FMU_V6C
HW type: V6C000001
HW version: 0x000
HW revision: 0x001
PX4 git-hash: 80171359656faf881dc643022f61ebde734d7cb6
PX4 version: Release 1.15.3 (17761279)
OS: NuttX
OS version: Release 11.0.0 (184549631)
OS git-hash: 5d74bc138955e6f010a38e0f87f34e9a9019aecc
Build datetime: Dec  4 2024 17:15:44
Build uri: localhost
Build variant: default
Toolchain: GNU GCC, 9.3.1 20200408 (release)
PX4GUID: 000600000000363835323133511900320019
MCU: STM32H7[4|5]xxx, rev. V
nsh> 

### Flight controller

Holybro Pixhawk 6C

### Vehicle type

Multicopter

### How are the different components wired up (including port information)

Holybro 915 Telemetry via Telem1

BetaFPV SuperD  2.4 via Telem2

### Additional context

#24188

dirksavage88, 2025-03-05:
I have replicated this issue as well. I am not sure where the handoff is b/w QGC and the px4 firmware version

mrpollo, 2025-07-21:
I have just confirmed this by flashing the latest beta from QGC and received the current v1.16.0-rc3. You might have run into an issue where CI didn't push the binaries to S3, but I can confirm that this is no longer a concern.