PX4/PX4-Autopilot #24381: Add in the ability to connect to NTRIP Servers for RTK corrections

DroneWuKong, 2025-02-20:
### Describe problem solved by the proposed feature

Currently there is no way to connect px4 to network corrections for RTK. This has been implemented in ardupilot for a while now. 

This would allow a user to connect to an ntrip server via local network or over the internet and receive corrections data. It would also eliminate the need to have a gps base plugged into the GCS. 

An implementation of this would get us another step closer to be able to map as easily as the one I shall not name. 

### Describe your preferred solution

I honestly have no idea how, I just know Ardu does it (but janky), DJI does it, Autel does it....It is the most convenient way to get geospatially accurate data that is well known and used through out the survey world. 

I'd like a menu that can save multiple different profiles of different networks, each one would have the needed information that needs to be entered to connect to the corrections network. once connected a menu will pop up that lists the "MountPoint" which the user will then select. once selected that context menu will disappear. 

What will appear out of the abyss is the menu that you just saw before the context menu popped up. but will have changed cause there will be a bunch of numbers in green that show aircraft position and reference station position in addition to the residual accuracies between the two. 

### Describe possible alternatives

Forget RTK in general and adopt PPK as the one to rule all in regards to aerial survey/mapping. As with PPK processing you don't need any live connections. you correct everything after. 

Another option would be PPP or satellite based corrections like Trimble with RTX/PP-RTX

### Additional context

#24188

dakejahl, 2025-02-20:
This isn't a PX4 problem but rather a QGC / Companion Computer problem. NTRIP servers just give you RTCM corrections in the same way that a base station would. PX4 can already handle RTCM over mavlink via [GPS_RTCM_DATA](https://mavlink.io/en/messages/common.html#GPS_RTCM_DATA). It looks like Ardupilot solves this in mavproxy. 

I just took a quick look at QGC and there's a PR in development for this feature, looks like someone picked it up again in December.
https://github.com/mavlink/qgroundcontrol/pull/9493

If you're looking for a companion computer solution checkout our PointOne Polaris client
https://github.com/ARK-Electronics/polaris-client-mavlink

DroneWuKong, 2025-02-21:
> This isn't a PX4 problem but rather a QGC / Companion Computer problem. NTRIP servers just give you RTCM corrections in the same way that a base station would. PX4 can already handle RTCM over mavlink via [GPS_RTCM_DATA](https://mavlink.io/en/messages/common.html#GPS_RTCM_DATA). It looks like Ardupilot solves this in mavproxy.
> 
> I just took a quick look at QGC and there's a PR in development for this feature, looks like someone picked it up again in December. [mavlink/qgroundcontrol#9493](https://github.com/mavlink/qgroundcontrol/pull/9493)
> 
> If you're looking for a companion computer solution checkout our PointOne Polaris client https://github.com/ARK-Electronics/polaris-client-mavlink

Sorry, wrong place for the post. But thank you for the heads up. I'll take a look at it. 

PointOne, I have yet to use your service. I'm heavy in the mapping side of things, you guys have quite a reach.