PX4/PX4-Autopilot #24382: [Bug] When building firmware "collect2: Error"

DroneWuKong, 2025-02-20:
### Describe the bug

When building firmware there is a 

"collect2: error: ld returned 1 exit status
ninja: build stopped: subcommand failed.
make: *** [Makefile:227: px4_fmu-v6c_default] Error 1"

No changes were made, pulling from 1.16.0 Alpha2 to build for Holybro Pixhawk 6C

### To Reproduce

via WSL building firmware off the 1.16.0 Alpha 2 release. 

just execute a "make px4_fmu-v6c_default"

### Expected behavior

builds firmware without issues

### Screenshot / Media

https://i.imgur.com/fxPHopy.png

### Software Version

HW arch: PX4_FMU_V6C
HW type: V6C000001
HW version: 0x000
HW revision: 0x001
PX4 git-hash: 80171359656faf881dc643022f61ebde734d7cb6
PX4 version: Release 1.15.3 (17761279)
OS: NuttX
OS version: Release 11.0.0 (184549631)
OS git-hash: 5d74bc138955e6f010a38e0f87f34e9a9019aecc
Build datetime: Dec  4 2024 17:15:44
Build uri: localhost
Build variant: default
Toolchain: GNU GCC, 9.3.1 20200408 (release)
PX4GUID: 000600000000363835323133511900320019
MCU: STM32H7[4|5]xxx, rev. V
nsh> 

### Flight controller

Holybro Pixhawk 6 C

### Vehicle type

Multicopter

### How are the different components wired up (including port information)

BetaFPV SuperD 2.4 on Telem 2

Holybro 915 telem on Telem1

dakejahl, 2025-02-20:
Clean build? Any inadvertent changes?

This module shouldn't be getting built with v6c target, the error should not be happening
[image]

[image]

DroneWuKong, 2025-02-21:
> Clean build? Any inadvertent changes?
> 
> This module shouldn't be getting built with v6c target, the error should not be happening [image]
> 
> [image]

Oh man, definitely a clean build.  There is a good chance that could have happened, I built it and enabled CRSF, I may have slipped a digit. I'll report back shortly.