PX4/PX4-Autopilot #24398: Ros2+px4 rover offboard mode

samirertt, 2025-02-22:
### Describe the bug

so am trying to use the rover r1 and trying to use it in offboard mode first i tried using [px4_msgs::msg::TrajectorySetpoint](https://github.com/PX4/PX4-Autopilot/blob/e12c3c00a4e658a0206735eb7402afbbb0242b20/msg/TrajectorySetpoint.msg) but this msg is not for rover i think its for copter so i started using [px4_msgs::msg::VehicleThrustSetpoint](https://github.com/PX4/PX4-Autopilot/blob/e12c3c00a4e658a0206735eb7402afbbb0242b20/msg/VehicleThrustSetpoint.msg) + [px4_msgs::msg::VehicleTorqueSetpoint](https://github.com/PX4/PX4-Autopilot/blob/e12c3c00a4e658a0206735eb7402afbbb0242b20/msg/VehicleTorqueSetpoint.msg) to be able to move the rover but am not able to move the rover.
my code is this what am i doing wrong ?

### To Reproduce

i have used this example

`/****************************************************************************
 *
 * Copyright 2020 PX4 Development Team. All rights reserved.
 *
 * Redistribution and use in source and binary forms, with or without
 * modification, are permitted provided that the following conditions are met:
 *
 * 1. Redistributions of source code must retain the above copyright notice, this
 * list of conditions and the following disclaimer.
 *
 * 2. Redistributions in binary form must reproduce the above copyright notice,
 * this list of conditions and the following disclaimer in the documentation
 * and/or other materials provided with the distribution.
 *
 * 3. Neither the name of the copyright holder nor the names of its contributors
 * may be used to endorse or promote products derived from this software without
 * specific prior written permission.
 *
 * THIS SOFTWARE IS PROVIDED BY THE COPYRIGHT HOLDERS AND CONTRIBUTORS "AS IS"
 * AND ANY EXPRESS OR IMPLIED WARRANTIES, INCLUDING, BUT NOT LIMITED TO, THE
 * IMPLIED WARRANTIES OF MERCHANTABILITY AND FITNESS FOR A PARTICULAR PURPOSE
 * ARE DISCLAIMED. IN NO EVENT SHALL THE COPYRIGHT HOLDER OR CONTRIBUTORS BE
 * LIABLE FOR ANY DIRECT, INDIRECT, INCIDENTAL, SPECIAL, EXEMPLARY, OR
 * CONSEQUENTIAL DAMAGES (INCLUDING, BUT NOT LIMITED TO, PROCUREMENT OF
 * SUBSTITUTE GOODS OR SERVICES; LOSS OF USE, DATA, OR PROFITS; OR BUSINESS
 * INTERRUPTION) HOWEVER CAUSED AND ON ANY THEORY OF LIABILITY, WHETHER IN
 * CONTRACT, STRICT LIABILITY, OR TORT (INCLUDING NEGLIGENCE OR OTHERWISE)
 * ARISING IN ANY WAY OUT OF THE USE OF THIS SOFTWARE, EVEN IF ADVISED OF THE
 * POSSIBILITY OF SUCH DAMAGE.
 *
 ****************************************************************************/

/**
 * @brief Offboard control example
 * @file offboard_control.cpp
 * @addtogroup examples
 * @author Mickey Cowden <info@cowden.tech>
 * @author Nuno Marques <nuno.marques@dronesolutions.io>
 */

#include <px4_msgs/msg/offboard_control_mode.hpp>
#include <px4_msgs/msg/trajectory_setpoint.hpp>
#include <px4_msgs/msg/vehicle_thrust_setpoint.hpp>
#include <px4_msgs/msg/vehicle_torque_setpoint.hpp>
#include <px4_msgs/msg/vehicle_command.hpp>
#include <px4_msgs/msg/vehicle_control_mode.hpp>
#include <rclcpp/rclcpp.hpp>
#include <stdint.h>

#include <chrono>
#include <iostream>

using namespace std::chrono;
using namespace std::chrono_literals;
using namespace px4_msgs::msg;

class OffboardControl : public rclcpp::Node
{
public:
	OffboardControl() : Node("offboard_control")
	{

		offboard_control_mode_publisher_ = this->create_publisher<OffboardControlMode>("/fmu/in/offboard_control_mode", 10);
		vehicle_thrust_setpoint_publisher_ = this->create_publisher<VehicleThrustSetpoint>("/fmu/in/vehicle_thrust_setpoint", 10);
		vehicle_torque_setpoint_publisher_ = this->create_publisher<VehicleTorqueSetpoint>("/fmu/in/vehicle_torque_setpoint", 10);
		vehicle_command_publisher_ = this->create_publisher<VehicleCommand>("/fmu/in/vehicle_command", 10);

		offboard_setpoint_counter_ = 0;

		auto timer_callback = [this]() -> void {

			if (offboard_setpoint_counter_ == 10) {
				// Change to Offboard mode after 10 setpoints
				this->publish_vehicle_command(VehicleCommand::VEHICLE_CMD_DO_SET_MODE, 1, 6);

				// Arm the vehicle
				this->arm();
			}

			// offboard_control_mode needs to be paired with trajectory_setpoint
			publish_offboard_control_mode();
			publish_vehicle_thrust_setpoint();
			publish_vehicle_torque_setpoint();

			// stop the counter after reaching 11
			if (offboard_setpoint_counter_ < 11) {
				offboard_setpoint_counter_++;
			}
		};
		timer_ = this->create_wall_timer(100ms, timer_callback);
	}

	void arm();
	void disarm();

private:
	rclcpp::TimerBase::SharedPtr timer_;

	rclcpp::Publisher<OffboardControlMode>::SharedPtr offboard_control_mode_publisher_;
	rclcpp::Publisher<VehicleCommand>::SharedPtr vehicle_command_publisher_;
	rclcpp::Publisher<VehicleThrustSetpoint>::SharedPtr vehicle_thrust_setpoint_publisher_;
	rclcpp::Publisher<VehicleTorqueSetpoint>::SharedPtr vehicle_torque_setpoint_publisher_;

	std::atomic<uint64_t> timestamp_;   //!< common synced timestamped

	uint64_t offboard_setpoint_counter_;   //!< counter for the number of setpoints sent

	void publish_offboard_control_mode();
	
	//void publish_trajectory_setpoint();
	void publish_vehicle_thrust_setpoint();
	void publish_vehicle_torque_setpoint();
	void publish_vehicle_command(uint16_t command, float param1 = 0.0, float param2 = 0.0);
};

/**
 * @brief Send a command to Arm the vehicle
 */
void OffboardControl::arm()
{
	publish_vehicle_command(VehicleCommand::VEHICLE_CMD_COMPONENT_ARM_DISARM, 1.0);

	RCLCPP_INFO(this->get_logger(), "Arm command send");
}

/**
 * @brief Send a command to Disarm the vehicle
 */
void OffboardControl::disarm()
{
	publish_vehicle_command(VehicleCommand::VEHICLE_CMD_COMPONENT_ARM_DISARM, 0.0);

	RCLCPP_INFO(this->get_logger(), "Disarm command send");
}

/**
 * @brief Publish the offboard control mode.
 *        For this example, only position and altitude controls are active.
 */
void OffboardControl::publish_offboard_control_mode()
{
	OffboardControlMode msg{};
	msg.position = false;
	msg.velocity = false;
	msg.acceleration = true;
	msg.attitude = false;
	msg.body_rate = false;
	msg.timestamp = this->get_clock()->now().nanoseconds() / 1000;
	offboard_control_mode_publisher_->publish(msg);
}

/**
 * @brief Publish a trajectory setpoint
 *        For this example, it sends a trajectory setpoint to make the
 *        vehicle hover at 5 meters with a yaw angle of 180 degrees.
 */
/*
void OffboardControl::publish_trajectory_setpoint()
{
	TrajectorySetpoint msg{};
	msg.position = {5.0, 0.0, 0.0}; // Move 5m forward, 3m right, stay on ground
	msg.yaw = 1.57; // Face right (90 degrees)
	msg.timestamp = this->get_clock()->now().nanoseconds() / 1000;
	trajectory_setpoint_publisher_->publish(msg);
}
*/

void OffboardControl::publish_vehicle_thrust_setpoint()
{
	VehicleThrustSetpoint msg{};
	msg.xyz = {1.0, 0.0, 0.0};
	msg.timestamp = this->get_clock()->now().nanoseconds() / 1000;
	vehicle_thrust_setpoint_publisher_->publish(msg);
}

void OffboardControl::publish_vehicle_torque_setpoint()
{
	VehicleTorqueSetpoint msg{};
	msg.xyz = {0.0, 0.0, 1.0};
	msg.timestamp = this->get_clock()->now().nanoseconds() / 1000;
	vehicle_torque_setpoint_publisher_->publish(msg);
}
/**
 * @brief Publish vehicle commands
 * @param command   Command code (matches VehicleCommand and MAVLink MAV_CMD codes)
 * @param param1    Command parameter 1
 * @param param2    Command parameter 2
 */
void OffboardControl::publish_vehicle_command(uint16_t command, float param1, float param2)
{
	VehicleCommand msg{};
	msg.param1 = param1;
	msg.param2 = param2;
	msg.command = command;
	msg.target_system = 1;
	msg.target_component = 1;
	msg.source_system = 1;
	msg.source_component = 1;
	msg.from_external = true;
	msg.timestamp = this->get_clock()->now().nanoseconds() / 1000;
	vehicle_command_publisher_->publish(msg);
}

int main(int argc, char *argv[])
{
	std::cout << "Starting offboard control node..." << std::endl;
	setvbuf(stdout, NULL, _IONBF, BUFSIZ);
	rclcpp::init(argc, argv);
	rclcpp::spin(std::make_shared<OffboardControl>());

	rclcpp::shutdown();
	return 0;
}`

### Expected behavior

any movement

### Software Version

latest version of px4 16beta

### Vehicle type

Rover

chfriedrich98, 2025-03-03:
A very recent PR (https://github.com/PX4/PX4-Autopilot/pull/24318) added offboard support for differential rovers through the `trajectroySetpoint.msg`.
To use this you need to build the `main` branch of PX4, since this feature is not yet part of any release.
Just FYI: There is no documentation yet since this is a very recent change but I am working on it.

samirertt, 2025-03-22:
thanks alot how could i know about the updates ? i am gonna try out trajectroySetpoint.msg. also what about the Ackermann version?

chfriedrich98, 2025-03-24:
@samirertt The ackermann module also supports the trajectorySetpoint.msg for now (was part of this [PR](https://github.com/PX4/PX4-Autopilot/pull/24285)).

These updates are usually documented in the [user guide](https://docs.px4.io/main/en/frames_rover/#rovers). However, offboard for rovers is  a WIP and I am currently working on directly exposing the rover messages (rate, attitude, velocity, position, throttle, steering, ... ) for offboard mode which will be much more straightforward to use than the trajectorySetpoint.msg.

In general if you are interested in the rover development, I try to keep the [Rover Project](https://github.com/orgs/PX4/projects/50) updated with the features that are being worked on.

leeeju, 2025-07-01:
@chfriedrich98 Hi. I am working on the coordination of a drone and a rover using px4, but I have succeeded in controlling the drone but I have not yet mastered how to actuate the simulation of the rover, I tried to do it the same way as the drone (trajectory_setpoint) but my rover did not actuate, what messages should I use to actuate the rover in the simulation? Optimally I would like to use Multiple Vehicles with ROS 2 and Gazebo to control the drone and the rover simultaneously, can you advise me?

leeeju, 2025-07-17:
What messages should I use to move my rover to a target point?

1. RoverPositionSetpoint
2. RoverVelocitySetpoint
3. TrajectorySetpoint
4. 
I'm testing now with RoverPositionSetpoint and there is no movement

closed 2026-01-31 by github-actions[bot]