PX4/PX4-Autopilot #24401: [Bug] Unable to calibrate ESC - told to disconnect battery despite no battery connected

Daidalos94, 2025-02-24:
### Describe the bug

As described, am using firmware 1.15.2 which is the latest on Qgroundcontrol auto firmware update (Although qground control states as 1.15.4 but upon upgrade, it became 1.15.2). Have also tried on a 1.15.0 firmware. When trying to calibrate my ESC, i was told to remove batteries as shown in the screenshot below. However, My pixhawk 6X and power module is not connected to any batteries. Have tried doing the same for a pixhawk 6X with firmware 1.14.3 and it allowed me to calibrate esc just fine on Qgroundcontrol. Anyone encountered the same issue? And any way to fix this besides going to an older firmware or manually calibrating my ESC?

### Screenshot / Media

[image]

### Software Version

1.15.2 

### Flight controller

Pixhawk 6X (holybro)

### Vehicle type

Multicopter

### How are the different components wired up (including port information)

Pixhawk 6X to PM07 holybro power module using ADS1115 ADC

YHuniv, 2025-03-31:
I got same error, I was often operating in Offboard mode in QGroundControl. Simply switching back to Manual mode allowed the ESC calibration to proceed normally.

The error message:

**"You must disconnect the battery prior to performing ESC Calibration. Disconnect your battery and try again."**

might be misleading in this context. A more appropriate message could be something like:

**"ESC calibration failed: please ensure you are in Manual mode."**

I hope this helps others facing the same issue.

MaEtUgR, 2025-05-06:
> I was often operating in Offboard mode in QGroundControl. Simply switching back to Manual mode allowed the ESC calibration to proceed normally.

There is no correlation between the navigation mode like Offboard and Manual and the ability to do an ESC calibration. I'd expect something else also changed in between these two tries e.g. the armed state or something related. Do you have a log of when that happened? You can either temporarily set `SDLOG_MODE` to log from boot and reboot to capture that or manually starts logging from the MAVLink shell while doing the calibration with `logger on` before and `logger off` after, see https://docs.px4.io/main/en/dev_log/logging.html

Mahmood-Hussain, 2025-05-23:
I encountered the same problem "Ares 6S 4IN1 32 BIT 128KHz Brushless ESC Speed Controller Blheli_32 6-layer" ESC. I tried with another drone it worked but for this the issue persists. 

[syslog_backup.log](https://github.com/user-attachments/files/20402770/syslog_backup.log)

Mahmood-Hussain, 2025-05-23:
For me it worked after connecting RC

Jimmy-Zhang13798, 2025-06-02:
Hi, in my case, I powered the whole system(FC, motors, ESCs) with a battery and connected the USB to my computer. Then on QGC, I went to the ESC calibration. and followed the pop-up instructions, which means that you should unplug the battery at this moment.

It should be able to solve this issue.

PR #24800 (2025-06-04): Fix reporting of connected battery
Fixes #24401
Fixes #24645
Fixes #24639

I haven't been able to verify this since I can't seem to build a firmware on Mac arm. But I believe this fixes the problem. Hopefully someone else can verify. And then this needs to get into v1.16 as well however that happens.