PX4/PX4-Autopilot #24408: Flash Test in CI is failing due to Github API

mrpollo, 2025-02-24:
### Describe the bug

The test for Flash space in the firmware is currently failing when posting a comment. The way the workflow was intended to work, CI would post a comment to the Pull Request with a summary of the flash space unavailable/available.  I have been trying to find the root cause but haven't been able to trace it down. 

IMO we should disable the commenting part of the workflow until we can find out a way to have this succeed to post.

**Relevant PRs**

* https://github.com/PX4/PX4-Autopilot/pull/24393
* https://github.com/PX4/PX4-Autopilot/pull/24404

### Vehicle type

None

alexcekay, 2025-02-25:
Hi @mrpollo:
I described the cause here: https://github.com/PX4/PX4-Autopilot/pull/24395#issuecomment-2678238157
Based on that the way to get it running on fork PRs is changing the target to `pull_request_target`, but that also changes the behavior as described in my comment.

By the way: Even if we don't directly find a fix for creating comments in case of forks we don't need to disable it completely as it still does work when it is not a fork and those are the majority of the PRs. If it bothers us that the job fails in cases of forks it is possible to add a condition to the comment part (rough idea from here: https://github.com/peter-evans/create-pull-request/blob/main/docs/concepts-guidelines.md#restrictions-on-repository-forks)

```
on: pull_request
jobs:
  example:
    runs-on: ubuntu-latest
    # Check if the event is not triggered by a fork
    if: github.event.pull_request.head.repo.full_name == github.repository
```

mrpollo, 2025-02-25:
Hey, @alexcekay, thanks for your help; we have been trying to get this fixed for a few weeks now and seem to be getting closer, but we haven't totally found the issue.

Regarding failing tests, I think they should either pass or fail. We currently have a big issue with devs ignoring CI tests, mostly because they are known to constantly fail, and we want to avoid this situation by having tests we know always work.

IMO, we should find the problem with the workflow permissions so we can post comments from Forks. We seem to be missing a very specific permission. I'll continue testing on another repo where we can debug faster. Once we have the fix, we can pull it into this workflow.

My plan is to get this workflow back to posting comments before we push the v1.16 release.

closed 2025-05-08 by dakejahl