PX4/PX4-Autopilot #24413: [Bug] Pixhawk v6c uXRCE-DDS-Client (PX4-ROS 2/DDS Bridge) don't connect with  uXRCE Agent using serial port and ROS2 tutorial

tuananh0606-tech, 2025-02-25:
### Describe the bug

I i got the same error as https://github.com/PX4/PX4-Autopilot/issues/21393 but using uXRCE-DDS (PX4-ROS 2/DDS Bridge), , i tried to fix it but it didn't work

### To Reproduce

1. This is my pixhawk 6c version
[image]
2. Installed UXRCE_DDS_CFG to telem2

[image]

3. I have set the parameters according to the path
https://docs.px4.io/main/en/peripherals/mavlink_peripherals.html

4. I also installed PX4-Autopilot version 1.15.2, installed additional firmware make px4_fmu-v6c_default and double checked the pixhawk 6c version in Nutx

[image]

[image]

### Expected behavior

I want the output to be the same as when connected to jmavsim simulation

[image]

### Screenshot / Media

but the result I got was not like that

[image]

and when restarting NuttX uxrce_dds_client is no longer running
[image]

### Vehicle type

None

beniaminopozzan, 2025-02-25:
Hi @tuananh0606-tech , could you share your entire parameter list please?

tuananh0606-tech, 2025-02-25:
edds

tuananh0606-tech, 2025-02-25:
Hi @beniaminopozzan , sorry you. I just pressed close by mistake. Can you help me, please?

tuananh0606-tech, 2025-02-26:
@beniaminopozzan  I switched to telem3 it works like that, but there is no publisher, when I tried to switch back to telem2 it still can't run.
```
[1740531532.837554] info     | TermiosAgentLinux.cpp | init                     | running...             | fd: 3
[1740531532.837720] info     | Root.cpp           | set_verbose_level        | logger setup           | verbose_level: 4
[1740531539.719673] info     | Root.cpp           | create_client            | create                 | client_key: 0x00000001, session_id: 0x81
[1740531539.719772] info     | SessionManager.hpp | establish_session        | session established    | client_key: 0x00000001, address: 1
[1740531540.007395] info     | ProxyClient.cpp    | create_participant       | participant created    | client_key: 0x00000001, participant_id: 0x001(1)
[1740531540.274864] info     | ProxyClient.cpp    | create_topic             | topic created          | client_key: 0x00000001, topic_id: 0x800(2), participant_id: 0x001(1)
[1740531540.275224] info     | ProxyClient.cpp    | create_subscriber        | subscriber created     | client_key: 0x00000001, subscriber_id: 0x800(4), participant_id: 0x001(1)
[1740531540.277025] info     | ProxyClient.cpp    | create_datareader        | datareader created     | client_key: 0x00000001, datareader_id: 0x800(6), subscriber_id: 0x800(4)
[1740531540.506152] info     | ProxyClient.cpp    | create_topic             | topic created          | client_key: 0x00000001, topic_id: 0x801(2), participant_id: 0x001(1)
[1740531540.506285] info     | ProxyClient.cpp    | create_subscriber        | subscriber created     | client_key: 0x00000001, subscriber_id: 0x801(4), participant_id: 0x001(1)
[1740531540.506990] info     | ProxyClient.cpp    | create_datareader        | datareader created     | client_key: 0x00000001, datareader_id: 0x801(6), subscriber_id: 0x801(4)
[1740531540.754290] info     | ProxyClient.cpp    | create_topic             | topic created          | client_key: 0x00000001, topic_id: 0x802(2), participant_id: 0x001(1)
[1740531540.754400] info     | ProxyClient.cpp    | create_subscriber        | subscriber created     | client_key: 0x00000001, subscriber_id: 0x802(4), participant_id: 0x001(1)
[1740531540.755268] info     | ProxyClient.cpp    | create_datareader        | datareader created     | client_key: 0x00000001, datareader_id: 0x802(6), subscriber_id: 0x802(4)
[1740531541.185578] info     | ProxyClient.cpp    | create_topic             | topic created          | client_key: 0x00000001, topic_id: 0x803(2), participant_id: 0x001(1)
[1740531541.185696] info     | ProxyClient.cpp    | create_subscriber        | subscriber created     | client_key: 0x00000001, subscriber_id: 0x803(4), participant_id: 0x001(1)
[1740531541.186456] info     | ProxyClient.cpp    | create_datareader        | datareader created     | client_key: 0x00000001, datareader_id: 0x803(6), subscriber_id: 0x803(4)
[1740531541.428461] info     | ProxyClient.cpp    | create_topic             | topic created          | client_key: 0x00000001, topic_id: 0x804(2), participant_id: 0x001(1)
[1740531541.428603] info     | ProxyClient.cpp    | create_subscriber        | subscriber created     | client_key: 0x00000001, subscriber_id: 0x804(4), participant_id: 0x001(1)
[1740531541.429367] info     | ProxyClient.cpp    | create_datareader        | datareader created     | client_key: 0x00000001, datareader_id: 0x804(6), subscriber_id: 0x804(4)
[1740531541.651400] info     | ProxyClient.cpp    | create_topic             | topic created          | client_key: 0x00000001, topic_id: 0x805(2), participant_id: 0x001(1)
[1740531541.651542] info     | ProxyClient.cpp    | create_subscriber        | subscriber created     | client_key: 0x00000001, subscriber_id: 0x805(4), participant_id: 0x001(1)
[1740531541.652262] info     | ProxyClient.cpp    | create_datareader        | datareader created     | client_key: 0x00000001, datareader_id: 0x805(6), subscriber_id: 0x805(4)
[1740531541.886531] info     | ProxyClient.cpp    | create_topic             | topic created          | client_key: 0x00000001, topic_id: 0x806(2), participant_id: 0x001(1)
[1740531541.886647] info     | ProxyClient.cpp    | create_subscriber        | subscriber created     | client_key: 0x00000001, subscriber_id: 0x806(4), participant_id: 0x001(1)
[1740531541.887380] info     | ProxyClient.cpp    | create_datareader        | datareader created     | client_key: 0x00000001, datareader_id: 0x806(6), subscriber_id: 0x806(4)
[1740531542.153602] info     | ProxyClient.cpp    | create_topic             | topic created          | client_key: 0x00000001, topic_id: 0x807(2), participant_id: 0x001(1)
[1740531542.153717] info     | ProxyClient.cpp    | create_subscriber        | subscriber created     | client_key: 0x00000001, subscriber_id: 0x807(4), participant_id: 0x001(1)
[1740531542.154417] info     | ProxyClient.cpp    | create_datareader        | datareader created     | client_key: 0x00000001, datareader_id: 0x807(6), subscriber_id: 0x807(4)
[1740531542.395934] info     | ProxyClient.cpp    | create_topic             | topic created          | client_key: 0x00000001, topic_id: 0x808(2), participant_id: 0x001(1)
[1740531542.396073] info     | ProxyClient.cpp    | create_subscriber        | subscriber created     | client_key: 0x00000001, subscriber_id: 0x808(4), participant_id: 0x001(1)
[... cut ...]
[1740531547.440567] info     | ProxyClient.cpp    | create_subscriber        | subscriber created     | client_key: 0x00000001, subscriber_id: 0x81C(4), participant_id: 0x001(1)
[1740531547.441376] info     | ProxyClient.cpp    | create_datareader        | datareader created     | client_key: 0x00000001, datareader_id: 0x81C(6), subscriber_id: 0x81C(4)
[1740531547.675540] info     | ProxyClient.cpp    | create_replier           | replier created        | client_key: 0x00000001, requester_id: 0x800(7), participant_id: 0x001(1)
```

tuananh0606-tech, 2025-02-26:
I tried searching for the topic but couldn't find it.
[image]

tuananh0606-tech, 2025-03-03:
@tassalor1 I disabled telem2 as instructed and uxrce dds client worked but when I ros2 topic list still no topic shows up

YHuniv, 2025-03-04:
I am facing the same problem.
The client is running, but not connected !!

`uxrce_dds_client start -t serial -d /dev/ttyS4 -b 921600
ERROR [uxrce_dds_client] Task already running
nsh> uxrce_dds_client status
INFO  [uxrce_dds_client] Running, disconnected
INFO  [uxrce_dds_client] Using transport:     serial
INFO  [uxrce_dds_client] timesync converged: false
uxrce_dds_client: cycle: 0 events, 0us elapsed, 0.00us avg, min 0us max 0us 0.000us rms
uxrce_dds_client: cycle interval: 0 events, 0.00us avg, min 0us max 0us 0.000us rms`

beniaminopozzan, 2025-03-05:
@tuananh0606-tech 
you need to provide way more infos:

- what companion computer are you using
- what OS are you using on the companion computer
- which ROS 2 version
- how did you install the agent
- which version of the agent did you install

YHuniv15, 2025-03-05:
I hope you don’t mind me jumping in—I’m in the same situation and figured I’d share my setup instead of opening a new issue.

Companion computer: PC Asus (RTX2070)
OS: **Ubuntu** **22.04**
ROS 2 version: **Humble**
Agent installation: Followed the instructions from [PX4’s ROS 2 user guide](https://docs.px4.io/main/en/ros2/user_guide.html)
Agent version: **2.4.2**
For reference, simulation with Gazebo is working fine on my end

tuananh0606-tech, 2025-03-06:
@beniaminopozzan 
I'm using:
1. Companion computer: 
Device name	DESKTOP-ST1PRMH
Processor	Intel(R) Core(TM) i7-10750H CPU @ 2.60GHz   2.59 GHz
Installed RAM	32.0 GB (31.8 GB usable)
Device ID	138FBECB-EE4F-4B21-A08D-C2D469D6A262
Product ID	00327-35933-06611-AAOEM
System type	64-bit operating system, x64-based processor
GPU: RTX2070 Ti

2. OS: Ubuntu 22.04 Linux
3. ROS2: Humble
4. Agent installation: Followed the instructions from [PX4’s ROS 2 user guide](https://docs.px4.io/main/en/ros2/user_guide.html)
5. Agent version: 2.4.2
6.PX4 firmware version: 1.15.2

I also tried to install other version for Agent (v2.4.2, v3.0.0) and PX4 firmware (v1.14.2, v1.14.4, v1.15.2, v1.15.4) but it's even not running

tuananh0606-tech, 2025-03-06:
@YHuniv Have you tried changing to a different telemetry?

tuananh0606-tech, 2025-03-06:
@beniaminopozzan 
here is what i ran and it shows on nutxx with telemetry baud rate 57600 and I can't use with higher baudrate

[image]

```
nsh> dmesg
 /etc/init.d/rc.board_sensors
bmi055_accel #0 on SPI bus 1 rotation 4
bmi055_gyro #0 on SPI bus 1 rotation 4
icm42688p #0 on SPI bus 1 rotation 6
ms5611 #0 on I2C bus 4 (external) address 0x77
ist8310 #0 on I2C bus 4 (external) address 0xC
WARN  [SPI_I2C] ist8310: no instance started (no device on bus?)
ekf2 [581:237]
Starting Main GPS on /dev/ttyS0
Starting UXRCE-DDS Client on /dev/ttyS3
INFO  [logger] logger started (mode=all)
INFO  [uavcan] Node ID 1, bitrate 1000000

NuttShell (NSH) NuttX-11.0.0
nsh> WARN  [health_and_arming_checks] Preflight Fail: Strong magnetic interference
INFO  [uxrce_dds_client] synchronized with time offset 0    us
INFO  [uxrce_dds_client] successfully created rt/fmu/out/failsafe_flags data writer, topic id: 80
INFO  [uxrce_dds_client] successfully created rt/fmu/out/position_setpoint_triplet data writer, topic id: 149
INFO  [uxrce_dds_client] successfully created rt/fmu/out/sensor_combined data writer, topic id: 168
INFO  [uxrce_dds_client] successfully created rt/fmu/out/timesync_status data writer, topic id: 191
INFO  [uxrce_dds_client] successfully created rt/fmu/out/vehicle_attitude data writer, topic id: 206
INFO  [uxrce_dds_client] successfully created rt/fmu/out/vehicle_control_mode data writer, topic id: 212
INFO  [uxrce_dds_client] successfully created rt/fmu/out/vehicle_local_position data writer, topic id: 219
INFO  [uxrce_dds_client] successfully created rt/fmu/out/vehicle_odometry data writer, topic id: 224
INFO  [uxrce_dds_client] successfully created rt/fmu/out/vehicle_status data writer, topic id: 229
WARN  [timesync] RTT too high for timesync: 510 ms
WARN  [timesync] RTT too high for timesync: 510 ms
WARN  [timesync] RTT too high for timesync: 677 ms
WARN  [timesync] RTT too high for timesync: 579 ms
WARN  [timesync] RTT too high for timesync: 876 ms
WARN  [timesync] RTT too high for timesync: 493 ms
WARN  [timesync] RTT too high for timesync: 625 ms
WARN  [timesync] RTT too high for timesync: 586 ms
WARN  [timesync] RTT too high for timesync: 483 ms
WARN  [timesync] RTT too high for timesync: 637 ms
WARN  [timesync] RTT too high for timesync: 532 ms
WARN  [timesync] RTT too high for timesync: 510 ms
WARN  [timesync] RTT too high for timesync: 546 ms
WARN  [timesync] RTT too high for timesync: 341 ms
WARN  [timesync] RTT too high for timesync: 510 ms
WARN  [timesync] RTT too high for timesync: 372 ms
WARN  [timesync] RTT too high for timesync: 766 ms
WARN  [timesync] RTT too high for timesync: 537 ms
WARN  [timesync] RTT too high for timesync: 763 ms
WARN  [timesync] RTT too high for timesync: 584 ms
WARN  [timesync] RTT too high for timesync: 556 ms
WARN  [timesync] RTT too high for timesync: 566 ms
WARN  [timesync] RTT too high for timesync: 584 ms
WARN  [timesync] RTT too high for timesync: 521 ms
WARN  [timesync] RTT too high for timesync: 457 ms
WARN  [timesync] RTT too high for timesync: 668 ms
WARN  [timesync] RTT too high for timesync: 341 ms
WARN  [timesync] RTT too high for timesync: 416 ms
WARN  [timesync] RTT too high for timesync: 656 ms
INFO  [uxrce_dds_client] No ping response, disconnecting
INFO  [uxrce_dds_client] synchronized with time offset 0    us
INFO  [uxrce_dds_client] successfully created rt/fmu/out/failsafe_flags data writer, topic id: 80
INFO  [uxrce_dds_client] successfully created rt/fmu/out/position_setpoint_triplet data writer, topic id: 149
INFO  [uxrce_dds_client] successfully created rt/fmu/out/sensor_combined data writer, topic id: 168
INFO  [uxrce_dds_client] successfully created rt/fmu/out/timesync_status data writer, topic id: 191
INFO  [uxrce_dds_client] successfully created rt/fmu/out/vehicle_attitude data writer, topic id: 206
INFO  [uxrce_dds_client] successfully created rt/fmu/out/vehicle_control_mode data writer, topic id: 212
INFO  [uxrce_dds_client] successfully created rt/fmu/out/vehicle_local_position data writer, topic id: 219
INFO  [uxrce_dds_client] successfully created rt/fmu/out/vehicle_odometry data writer, topic id: 224
INFO  [uxrce_dds_client] successfully created rt/fmu/out/vehicle_status data writer, topic id: 229
```

tuananh0606-tech, 2025-03-06:
```
hta@g5:~$ sudo MicroXRCEAgent serial --dev /dev/ttyUSB0 -b 57600
[sudo] password for hta: 
[1741232028.234126] info     | TermiosAgentLinux.cpp | init                     | running...             | fd: 3
[1741232028.234279] info     | Root.cpp           | set_verbose_level        | logger setup           | verbose_level: 4
[1741232029.082628] info     | Root.cpp           | create_client            | create                 | client_key: 0x00000001, session_id: 0x81
[1741232029.091608] info     | SessionManager.hpp | establish_session        | session established    | client_key: 0x00000001, address: 1
[1741232029.361304] info     | ProxyClient.cpp    | create_participant       | participant created    | client_key: 0x00000001, participant_id: 0x001(1)
[1741232029.624643] info     | ProxyClient.cpp    | create_topic             | topic created          | client_key: 0x00000001, topic_id: 0x3E8(2), participant_id: 0x001(1)
[1741232029.624827] info     | ProxyClient.cpp    | create_subscriber        | subscriber created     | client_key: 0x00000001, subscriber_id: 0x3E8(4), participant_id: 0x001(1)
[1741232029.626452] info     | ProxyClient.cpp    | create_datareader        | datareader created     | client_key: 0x00000001, datareader_id: 0x3E8(6), subscriber_id: 0x3E8(4)
[1741232029.876605] info     | ProxyClient.cpp    | create_topic             | topic created          | client_key: 0x00000001, topic_id: 0x3E9(2), participant_id: 0x001(1)
[1741232029.876718] info     | ProxyClient.cpp    | create_subscriber        | subscriber created     | client_key: 0x00000001, subscriber_id: 0x3E9(4), participant_id: 0x001(1)
[1741232029.877457] info     | ProxyClient.cpp    | create_datareader        | datareader created     | client_key: 0x00000001, datareader_id: 0x3E9(6), subscriber_id: 0x3E9(4)
[1741232030.113264] info     | ProxyClient.cpp    | create_topic             | topic created          | client_key: 0x00000001, topic_id: 0x3EA(2), participant_id: 0x001(1)
[1741232030.113406] info     | ProxyClient.cpp    | create_subscriber        | subscriber created     | client_key: 0x00000001, subscriber_id: 0x3EA(4), participant_id: 0x001(1)
[1741232030.114201] info     | ProxyClient.cpp    | create_datareader        | datareader created     | client_key: 0x00000001, datareader_id: 0x3EA(6), subscriber_id: 0x3EA(4)
[1741232030.534442] info     | ProxyClient.cpp    | create_topic             | topic created          | client_key: 0x00000001, topic_id: 0x3EB(2), participant_id: 0x001(1)
[1741232030.534635] info     | ProxyClient.cpp    | create_subscriber        | subscriber created     | client_key: 0x00000001, subscriber_id: 0x3EB(4), participant_id: 0x001(1)
[1741232030.535411] info     | ProxyClient.cpp    | create_datareader        | datareader created     | client_key: 0x00000001, datareader_id: 0x3EB(6), subscriber_id: 0x3EB(4)
[1741232030.817730] info     | ProxyClient.cpp    | create_topic             | topic created          | client_key: 0x00000001, topic_id: 0x3EC(2), participant_id: 0x001(1)
[1741232030.817855] info     | ProxyClient.cpp    | create_subscriber        | subscriber created     | client_key: 0x00000001, subscriber_id: 0x3EC(4), participant_id: 0x001(1)
[1741232030.818624] info     | ProxyClient.cpp    | create_datareader        | datareader created     | client_key: 0x00000001, datareader_id: 0x3EC(6), subscriber_id: 0x3EC(4)
[1741232031.017783] info     | ProxyClient.cpp    | create_topic             | topic created          | client_key: 0x00000001, topic_id: 0x3ED(2), participant_id: 0x001(1)
[1741232031.017937] info     | ProxyClient.cpp    | create_subscriber        | subscriber created     | client_key: 0x00000001, subscriber_id: 0x3ED(4), participant_id: 0x001(1)
[1741232031.018713] info     | ProxyClient.cpp    | create_datareader        | datareader created     | client_key: 0x00000001, datareader_id: 0x3ED(6), subscriber_id: 0x3ED(4)
[1741232031.236159] info     | ProxyClient.cpp    | create_topic             | topic created          | client_key: 0x00000001, topic_id: 0x3EE(2), participant_id: 0x001(1)
[1741232031.236317] info     | ProxyClient.cpp    | create_subscriber        | subscriber created     | client_key: 0x00000001, subscriber_id: 0x3EE(4), participant_id: 0x001(1)
[1741232031.237136] info     | ProxyClient.cpp    | create_datareader        | datareader created     | client_key: 0x00000001, datareader_id: 0x3EE(6), subscriber_id: 0x3EE(4)
[1741232031.492408] info     | ProxyClient.cpp    | create_topic             | topic created          | client_key: 0x00000001, topic_id: 0x3EF(2), participant_id: 0x001(1)
[1741232031.492602] info     | ProxyClient.cpp    | create_subscriber        | subscriber created     | client_key: 0x00000001, subscriber_id: 0x3EF(4), participant_id: 0x001(1)
[1741232031.493405] info     | ProxyClient.cpp    | create_datareader        | datareader created     | client_key: 0x00000001, datareader_id: 0x3EF(6), subscriber_id: 0x3EF(4)
[1741232031.696774] info     | ProxyClient.cpp    | create_topic             | topic created          | client_key: 0x00000001, topic_id: 0x3F0(2), participant_id: 0x001(1)
[1741232031.696937] info     | ProxyClient.cpp    | create_subscriber        | subscriber created     | client_key: 0x00000001, subscriber_id: 0x3F0(4), participant_id: 0x001(1)
[... cut ...]
[1741232355.891989] info     | ProxyClient.cpp    | create_topic             | topic created          | client_key: 0x00000001, topic_id: 0x0E5(2), participant_id: 0x001(1)
[1741232355.892099] info     | ProxyClient.cpp    | create_publisher         | publisher created      | client_key: 0x00000001, publisher_id: 0x0E5(3), participant_id: 0x001(1)
[1741232355.892453] info     | ProxyClient.cpp    | create_datawriter        | datawriter created     | client_key: 0x00000001, datawriter_id: 0x0E5(5), publisher_id: 0x0E5(3)
```

beniaminopozzan, 2025-03-06:
Ah you are using a usb2 serial adapter @tuananh0606-tech 

> Agent version: 2.4.2
PX4 firmware version: 1.15.2

perfect, these pair is compatible

> telemetry baud rate 57600 and I can't use with higher baudrate

Why not?

YHuniv, 2025-03-06:
Yes, MAVlink is disabled on TELEM2.

beniaminopozzan, 2025-03-06:
were you running uxrcedds over a telemetry radio?!

>To complete, I am not sur but maybe the probleme comes from my
[Holybro SiK Telemetry Radio V3](https://holybro.com/products/sik-telemetry-radio-v3?srsltid=AfmBOooz8pCNnW7ClR-I06RjyeUhZsbziuEKFqXobUMOKbS1qVOrzjEu) connected to Telem2 of my Pixhawk 6X.

Ok, you kind of wrote that but you never really wrote that you were using TELEM 2 for uxrcedds....

tassalor1, 2025-03-06:
This seems to be a problem with the UXRCE agent. I did exactly as the docs say to install on Ubuntu 20.04 on my nano. Never got it to work. I had to downgrade to v1.3 and use fast RPS instead, it now work perfectly fine now

invidia0, 2025-03-11:
Hi all, I was having kinda the same issue where after following [ROS 2 User Guide
](https://docs.px4.io/main/en/ros2/user_guide.html) I could not see any topic with `ros2 topic list`. I tried different combination of uXRCE-DDS, PX4 and found a working solution with the following setup:
- ROS 2 Humble
- Micro-XRCE-DDS-Agent: `v2.4.2` with the FastDDS PR [here](https://github.com/eProsima/Micro-XRCE-DDS-Agent/pull/383#issue-2899821803) in order to fix the 2.12.x problem during build (just switch to 2.14).
**Note:** I also tried with the latest 3.0.0 version but it didn't work in my case.
- PX4: `v1.15.4`
- px4_msgs: `release/1.15`
- px4_ros_comm: `release/1.15`

With this setup I succesfully see topics and can run the example code in the guide:

`ros2 launch px4_ros_com sensor_combined_listener.launch.py`

Hope this helps!

tassalor1, 2025-03-23:
@invidia0 is this with jetson nano also? I've tried what you did with my foxy jetson nano and still can't connect to DDs agent

tuananh0606-tech, 2025-04-24:
https://docs.px4.io/main/en/companion_computer/pixhawk_rpi.html
Hey guys, we noticed that uXRCE-DDS can only establish a stable connection at high baud rates. Because of this, we had to use the TELEM2 port on the Pixhawk to connect to the companion computer. This way, we're able to successfully receive ROS 2 topics published by PX4. Thanks for looking into this topic.

closed 2025-04-24 by tuananh0606-tech

PeterKarstensen, 2025-05-23:
I am facing the same problems, however, following the steps in the thread here have not solved the problem for me. 

I use a Pixhawk v6c and my setup is:
- ROS 2 Humble
- Micro-XRCE-DDS-Agent: v2.4.2 with the FastDDS, where I used FAST DDS v2.14.3
- PX4: v1.15.4

I use TELEM2 with baudrate 921600 via an FTDI.

This is the output in the MAVLink console on QGC:

`nsh: sysinit: fopen failed: No such file or directory

NuttShell (NSH) NuttX-11.0.0
nsh> dmesg
HW arch: PX4_FMU_V6C
HW type: V6C000001
HW version: 0x000
HW revision: 0x001
PX4 git-hash: 99c40407ffd7ac184e2d7b4b293f36f10fe561ef
PX4 version: Release 1.15.4 (17761535)
OS: NuttX
OS version: Release 11.0.0 (184549631)
OS git-hash: 5d74bc138955e6f010a38e0f87f34e9a9019aecc
Build datetime: Mar 20 2025 23:02:15
Build uri: localhost
Build variant: default
Toolchain: GNU GCC, 9.3.1 20200408 (release)
PX4GUID: 000600000000383033343430510600400027
MCU: STM32H7[4|5]xxx, rev. V
INFO  [param] selected parameter default file /fs/mtd_params
INFO  [param] importing from '/fs/mtd_params'
INFO  [parameters] BSON document size 2085 bytes, decoded 2085 bytes (INT32:38, FLOAT:63)
INFO  [param] selected parameter backup file /fs/microsd/parameters_backup.bson
Board architecture defaults: /etc/init.d/rc.board_arch_defaults
Board defaults: /etc/init.d/rc.board_defaults
Loading airframe: /etc/init.d/airframes/4019_x500_v2
INFO  [dataman] data manager file '/fs/microsd/dataman' size is 68528 bytes
rgbled_ncp5623c #0 on I2C bus 1 address 0x39
INFO  [px4io] IO FW CRC match
Board sensors: /etc/init.d/rc.board_sensors
bmi055_accel #0 on SPI bus 1 rotation 4
bmi055_gyro #0 on SPI bus 1 rotation 4
icm42688p #0 on SPI bus 1 rotation 6
ms5611 #0 on I2C bus 4 (external) address 0x77
ist8310 #0 on I2C bus 4 (external) address 0xC
ist8310 #1 on I2C bus 1 (external) address 0xE rotation 10
WARN  [SPI_I2C] Already running on bus 1
WARN  [vehicle_angular_velocity] no gyro selected, using sensor_gyro_fifo:0 4325386
ekf2 [743:237]
Starting Main GPS on /dev/ttyS0
Starting UXRCE-DDS Client on /dev/ttyS3
INFO  [uxrce_dds_client] init serial /dev/ttyS3 @ 921600 baud
INFO  [cdcacm_autostart] Starting CDC/ACM autostart
INFO  [logger] logger started (mode=all)
INFO  [uavcan] Node ID 1, bitrate 1000000

NuttShell (NSH) NuttX-11.0.0
nsh> sercon: Registering CDC/ACM serial driver
sercon: Successfully registered the CDC/ACM serial driver
INFO  [cdcacm_autostart] Starting mavlink on /dev/ttyACM0 (SYS_USB_AUTO=2)
INFO  [mavlink] mode: Onboard, data rate: 100000 B/s on /dev/ttyACM0 @ 2000000B
uavcan_mag adding channel for topic sensor_mag node 123...
uavcan_mag channel 123 instance 2 ok
INFO  [mavlink] Starting mavlink shell
nsh> ps
  PID GROUP PRI POLICY   TYPE    NPX STATE    EVENT     SIGMASK   STACK   USED  FILLED COMMAND
    0     0   0 FIFO     Kthread N-- Ready              00000000 000726 000412  56.7%  Idle Task
    1     1 249 FIFO     Kthread --- Waiting  Semaphore 00000000 001224 000292  23.8%  hpwork 0x24000998
    2     2  50 FIFO     Kthread --- Waiting  Semaphore 00000000 001576 000292  18.5%  lpwork 0x240009ac
[... 12 lines cut ...]
  976   976 100 FIFO     Task    --- Waiting  Semaphore 00000000 009872 007780  78.8%  uxrce_dds_client start -t serial -d /dev/ttyS3 -b p:SER_TEL2_BAUD
  881   881 205 FIFO     Task    --- Waiting  Semaphore 00000000 001936 001204  62.1%  gps start -d /dev/ttyS0 -b p:SER_GPS1_BAUD
 1074  1074 100 FIFO     Task    --- Running            00000000 002000 001492  74.6%  mavlink_shell
 1043  1043 230 FIFO     Task    --- Waiting  Semaphore 00000000 003616 003060  84.6%! logger start -b 64 -t
 1045  1043  60 FIFO     pthread --- Waiting  Semaphore 00000000 001144 000388  33.9%  log_writer_file 0x300084a0
  982   982 105 FIFO     Task    --- Waiting  Semaphore 00000000 002104 001512  71.8%  navigator start
  727     4 241 FIFO     pthread --- Waiting  Semaphore 00000000 005976 003660  61.2%  wq:INS0 0x81f4354
  504     4 243 FIFO     pthread --- Waiting  Semaphore 00000000 002312 000996  43.0%  wq:I2C4 0x81f434c
  730     4 240 FIFO     pthread --- Waiting  Semaphore 00000000 005976 003660  61.2%  wq:INS1 0x81f435c
  732   732 140 FIFO     Task    --- Waiting  Signal    00000000 003192 001500  46.9%  commander start
 1053     4 236 FIFO     pthread --- Waiting  Semaphore 00000000 003600 003148  87.4%! wq:uavcan 0x81994a8
nsh> uxrce_dds_client status
INFO  [uxrce_dds_client] Running, disconnected
INFO  [uxrce_dds_client] Using transport:     serial
INFO  [uxrce_dds_client] timesync converged: false
uxrce_dds_client: cycle: 0 events, 0us elapsed, 0.00us avg, min 0us max 0us 0.000us rms
uxrce_dd`

On the my computer, when running the Agent it does not create the topics:

`pihka@pihka-HP-EliteBook-840-G8-Notebook-PC:~/git$ sudo MicroXRCEAgent serial --dev /dev/ttyUSB0 -b 921600 -v6
[1747990155.091143] info     | TermiosAgentLinux.cpp | init                     | running...             | fd: 3
[1747990155.091339] info     | Root.cpp           | set_verbose_level        | logger setup           | verbose_level: 6

`