PX4/PX4-Autopilot #24426: [Bug] setting WV_EN to enable and executing MAV_CMD_DO_VTOL_TRANSITION results in an error.

bonobo001, 2025-02-28:
### Describe the bug

I am operating a VTOL based on PX4 firmware version 1.15.2 or later. After setting WV_EN to enable and executing MAV_CMD_DO_VTOL_TRANSITION to transition from multicopter mode to fixed-wing mode, the following issues occurred:

1. Mission failure: Unable to reach heading within timeout.
2. After the mission failure, the vehicle did not transition to fixed-wing mode and remained in multicopter mode.

### To Reproduce

1. Set up a VTOL running PX4 firmware 1.15.2 or later.
2. Enable the WV_EN parameter.
3. Execute a VTOL transition mission:
  3.1. Perform a back transition (fixed-wing to multicopter).
  3.2. Then execute another transition to fixed-wing mode.

### Expected behavior

4. Issue observed:
  4.1. The aircraft's nose is not aligned with the next mission waypoint direction.
  4.2. The same "Mission failure: unable to reach heading within timeout" error occurs.
  4.3. The nose direction does not change as expected during the transition.

5. Additional findings:
  5.1. When the WV_EN parameter was disabled, the issue did not occur, and the transition proceeded as expected.
  5.2. In version 1.14.0, the logic was designed to control yaw during transition, even when the weather vane (WV_EN) feature was enabled. However, starting from version 1.15.0, some logic statements have been removed or modified.

6. Identified Code Changes
  6.1. Mission.cpp → Logic has been removed or modified.
     The following lines have been removed:
     pos_sp_triplet->current.disable_weather_vane = true;
     pos_sp_triplet->current.disable_weather_vane = false;

  6.2. PositionSetpoint.msg → The message has been removed.
     The following statement has been removed:
     bool disable_weather_vane

  6.3. FlightTaskAuto.cpp → The code has been modified.
     The following statement has been modified:
     _weathervane.setNavigatorForceDisabled(_sub_triplet_setpoint.get().current.disable_weather_vane);

* After applying the above modifications, MAV_CMD_DO_VTOL_TRANSITION successfully executes the transition even when WV_EN is enabled on PX4 version 1.15.0 or later.

### Screenshot / Media

[image]

### Flight Log

https://logs.px4.io/plot_app?log=1db2b97a-d16c-4097-a82a-be04ed17c649

### Software Version

PX4 v1.15.2 Ver.

### Flight controller

Holybro X6, SITL

### Vehicle type

Hybrid VTOL

bonobo001, 2025-05-26:
If possible, I would appreciate it transitioning towards the currently active waypoint. Thank you!

sfuhrer, 2025-06-13:
Makes sense, agree. Would you be up for proposing the fix? That would be very appreciated! Not sure how many other users have the same problem, as most VTOL users only transition during takeoff/landing phases.

closed 2026-01-19 by github-actions[bot]

marc-seninge, 2026-10-01:
This issue can be fixed in the current main branch without reintroducing disable_weather_vane.

In Mission::handleVtolTransition(), the "_mission_item" is correctly computed :

https://github.com/PX4/PX4-Autopilot/blob/27d3a62df1b457fb29bb899e368efdc572b371f7/src/modules/navigator/mission.cpp#L449 

but immediately afterwards mission_item_to_position_setpoint() does not use the computed "_mission_item" :

https://github.com/PX4/PX4-Autopilot/blob/27d3a62df1b457fb29bb899e368efdc572b371f7/src/modules/navigator/mission.cpp#L452

Therefore the yaw setpoint is not modified, the weather vane remains active and the drone does not align itself to the yaw expected by "_mission_item". This triggers the heading time-out :
https://github.com/PX4/PX4-Autopilot/blob/27d3a62df1b457fb29bb899e368efdc572b371f7/src/modules/navigator/mission_block.cpp#L421 

Proposed fix:
               mission_item_to_position_setpoint(_mission_item, &pos_sp_triplet->current);

I successfully tested this patch with "make px4_sitl gz_standard_vtol_windy" and the attached mission. 
Without my patch, this mission triggers the heading time-out.

The patch was applied and tested on PX4 commit 27d3a62df1b457fb29bb899e368efdc572b371f7 (git rev-parse HEAD).
[mission_votl_transition_wv_ko.txt](https://github.com/user-attachments/files/32919385/mission_votl_transition_wv_ko.txt)
[windy.sdf.txt](https://github.com/user-attachments/files/32920404/windy.sdf.txt)