PX4/PX4-Autopilot #24439: [Bug] VOXL2 Upgrade to 1.16 resets EV_CTRL parameter

afwilkin, 2025-03-03:
### Describe the bug

https://forum.modalai.com/topic/4156/px4-1-16-no-local-position

When you upgrade between PX4 ver. 1.14 or 1.15 to 1.16 the vehicle doesn't get a valid local position.
EV_CTRL changed from 15 in 1.15 and earlier to 0 in 1.16. Either check this parameter is set to 15 or run "voxl-configure-px4-params -p MRB-D0006" for Starling V1 or "voxl-configure-px4-params -p MRB-D0005-V2" for Starling V2.

### To Reproduce

Upgrade PX4 on VOXL2 starling to 1.16 release

### Expected behavior

Expected behavior is that this parameter stays the same. Stays at 15. The ACTUAL behavior is that it is reset to 0 and gives a "No valid local position" when trying to fly the drone.

### Flight Log

Since it flies fine after manually setting the parameter there is no follow up log 

### Software Version

1.16

### Flight controller

VOXL2

### Vehicle type

Multicopter

beniaminopozzan, 2025-03-27:
For standard boards you would normally add a `rc.board_defaults` in the board _init_ folder (https://github.com/PX4/PX4-Autopilot/blob/main/boards/holybro/kakuteh7/init/rc.board_defaults as an example) to set board-specific default parameters.

However, the `modalai_voxl` board does not use the usual startup procedure. The PX4 entrypoint for that board: `voxl_px4` do not call the `rcS` script therefore it is not able to run any of the `rc.*` configuration scripts.

Modalai board is heavily offboarding parameter configuration from PX4 main code and it primarily uses https://gitlab.com/voxl-public/voxl-sdk/utilities/voxl-px4-params/-/blob/master/scripts/voxl-configure-px4-params?ref_type=heads. Changing that just for a single parameter seems too an overkill in my opinion. Modalai users could just be pointed by the documentation to the importance of running `voxl-configure-px4-params` after flashing 1.16.

katzfey, 2025-04-01:
Thanks! I'll add an explicit set of the default to 15 for VOXL 2 board. See: https://github.com/PX4/PX4-Autopilot/pull/24648

closed 2026-01-19 by github-actions[bot]