PX4/PX4-Autopilot #24446: CSRF & GHST RC Protocol Support

afwilkin, 2025-03-04:
### Describe problem solved by the proposed feature

It is rather non-obvious how to setup CSRF RC support, and even harder to setup GHST RC. 

[Loom Video Overview](https://www.loom.com/share/84427d8002764c369b716ae674a62185?sid=28c5dd54-ab64-4b5b-b896-d5e0832b1a4b)

This module works on VOXL2, but on a standard Pixhawk6c build this module is not setup, and documentation is rather lacking

### Describe your preferred solution

Have this as part of the default build. 

Especially with GHST offering an NDAA solution, this option will only get more popular.

### Describe possible alternatives

Better documentation on how to get the CSRF & GHST protocols working.

dakejahl, 2025-03-04:
@patrikpordi did you ever figure out the GHST setup?

patrikpordi, 2025-03-04:
I did, for me it was pretty straight forward from the PX4 side, not from the Ghost side though. In PX4 you just need to enable `RC_GHST_PRT_CFG,` you have to select the port where you have connected the receiver, you should also disable the other RC variants. After that you can use `ghst_rc` status to see if it is on. Problems come from the Ghost side they ask you to connect their serial to your TX, which just won't work with PX4, so I ended up connecting to RX. I ended up having the connection but the RC inputs were just jumping all over without me touching it(I suspect faulty hardware from their side). Also you need to use their IOS/Android app to select the protocol you want to use, on the top of the app being pretty bad, the led combinations are not documented. They also suggest using SBus, I got the same result with that.

https://github.com/user-attachments/assets/f649b670-7d30-437c-966e-4d462d6ff67a

afwilkin, 2025-03-05:
Good to know that it works. probably just need to document how to enable it

patrikpordi, 2025-03-05:
At which stage are you getting stuck?

dirksavage88, 2025-03-05:
I believe getting the common (standalone) rc drivers default on px4 fmu's are blocked by https://github.com/PX4/PX4-Autopilot/pull/23428

hamishwillee, 2025-03-27:
For CRSF we have this doc https://docs.px4.io/main/en/telemetry/crsf_telemetry.html - is this what you mean by "CSRF RC support"?
The section on [PX4 Config](https://docs.px4.io/main/en/telemetry/crsf_telemetry.html#px4-configuration) seems reasonable to me - it covers the fact you need a dedicated build for that module, and what ports you need to set etc.

> nd even harder to setup GHST RC

I imagine so. Never heard of it. Would be good if one of you who has used it starts a doc modelled off https://docs.px4.io/main/en/telemetry/crsf_telemetry.html. I can then help polish it.

hamishwillee, 2025-06-05:
I'm trying to find out the "full story" in #24857. 

We could document the GHST story now because the rc_input and the IO board RC don't support GHST.

hamishwillee, 2026-01-29:
Oh Bot, what would we do without you.

hamishwillee, 2026-08-31:
I'm keeping this open. I have a plan. Its somewhere in my priorities.