PX4/PX4-Autopilot #24449: [Bug] PX4 Min PWM & Disarm PWM set to same values

afwilkin, 2025-03-05:
### Describe the bug

After I reset all parameters to firmware defaults, select the Holybro x500 airframe, configure everything, and takeoff in stabilized flight mode, the propellers do not spin when armed. Only when we start to throttle up do they start to spin, which could be a bit disorienting for someone trying to bring up a new airframe. 

### To Reproduce

1. Reflash PX4
2. Reset all firmware defaults in QGC (Parameters->Tools-> Reset to all firmware defaults)
3.  Go through QGC and configure all red tabs (calibrate sensors, assign Flight modes, calibrate RC, etc...)
4. Flip the drone into stabilized flight mode and arm. The motors shouldnt spin

### Expected behavior

Motors spin on arm

### Screenshot / Media

https://www.loom.com/share/f4d51a22a41f447e8565884861672368?sid=35965a9c-a580-4c99-87f3-3da5502c078e

[image]

### Software Version

1.16 (but I also see this happening in 1.15)

### Flight controller

Pixhawk 6c

### Vehicle type

None

### How are the different components wired up (including port information)

pwm escs

afwilkin, 2025-03-05:
Going to try with the generic layout, but this was for the holybro x500...which is pretty widely used

afwilkin, 2025-03-05:
Not an issue when selecting generic quadcopter as you are forced to select motor configurations which automatically get set to 

PWM Min: 1100
PWM Max: 1900

afwilkin, 2025-03-05:
We decided that we should take out those default functions for the x500 airframe, and investigate what other airframes this happens to .

we might want to SPECIFY that this is the ready to fly kit and not the almost ready to fly kit

MaEtUgR, 2025-03-05:
@afwilkin Thanks for pointing this out! That's definitely something others run into.
The issue is that the PWM function must not be set without also configuring the correct PWM minimum and maximum values. This is wrong here: https://github.com/PX4/PX4-Autopilot/blob/0ab3e45c13b603eaf867b7a341ab0eeafc4c92aa/ROMFS/px4fmu_common/init.d/airframes/4019_x500_v2#L25-L28

Discussion in the dev call was:
X500 frame is close to QuadX default. Short term we should remove the Motor mappings. But to have a useful X500 airframe we should make it very clear that it's supposed to only be used with the ARF https://holybro.com/products/x500-v2-kits setup which is wired according to a guide and then the specific configuration makes sense. But if someone uses the X500 frame with random motors, ESCs, hopefully DSHOT he's better off using the generic QuadX as a start.

afwilkin, 2025-03-05:
Deleted the lines and tested. Can confirm the Actuators tab becomes red and requires user to setup the correct motor mappings. 

[image]

Tested using the same steps here

Reflash PX4
Reset all firmware defaults in QGC (Parameters->Tools-> Reset to all firmware defaults)
Go through QGC and configure all red tabs (calibrate sensors, assign Flight modes, calibrate RC, etc...)
Flip the drone into stabilized flight mode and arm. The motors shouldnt spin

and the motors now spin when they arm. 

PR: https://github.com/PX4/PX4-Autopilot/pull/24459

afwilkin, 2025-03-05:
I dont want to completely close this because there are other airframes that set this as well

[image]

@MaEtUgR I am unsure if we should unset all of these...or just wait until someone complains and then do it. I do not have all of the airframes that are listed here

hamishwillee, 2025-03-06:
@MaEtUgR Unless I'm missing something fundamental, the right approach is to add appropriate limits for the assigned PWM - see https://github.com/PX4/PX4-Autopilot/pull/24459#issuecomment-2702505755

closed 2025-04-01 by afwilkin