PX4/PX4-Autopilot #24466: [Bug] JSBSim SITL crashes immediately after arming or after takeoff

simfinite, 2025-03-06:
### Describe the bug

Hi,

I am trying to get JSBSim SITL with the Rascal model to run [as documented here](https://docs.px4.io/main/en/sim_jsbsim/index.html). 

After applying a work-around for a [known problem with the JSBSim-Bridge](https://github.com/Auterion/px4-jsbsim-bridge/pull/66), I can run the simulation and connect with a QGCS. However, right after arming, the plane rolls over and crashes. 

I am working with latest main (have tested on 15.3 as well) on Ubuntu 24.04.

What am I missing? Can anyone reproduce this issue? Happy for any help!

Here is a full terminal log:

```
~/Code/PX4-Autopilot$ make px4_sitl jsbsim
[0/4] Performing build step for 'jsbsim_bridge'
ninja: no work to do.
[3/4] cd /home/user/Code/PX4-Autopilot/build/px4_sitl_default/src/modules/simulati...home/simon/Code/PX4-Autopilot /home/user/Code/PX4-Autopilot/build/px4_sitl_default
SITL ARGS
sitl_bin: /home/user/Code/PX4-Autopilot/build/px4_sitl_default/bin/px4
model: rascal
world: LSZH
src_path: /home/user/Code/PX4-Autopilot
build_path: /home/user/Code/PX4-Autopilot/build/px4_sitl_default
SITL COMMAND: "/home/user/Code/PX4-Autopilot/build/px4_sitl_default/bin/px4" "/home/user/Code/PX4-Autopilot/build/px4_sitl_default"/etc

     JSBSim Flight Dynamics Model v1.2.1 [GitHub build 1348/commit 9b95d1b5ccff59916c79a0e3eb8f548377910598] Aug  8 2024 18:52:06
            [JSBSim-ML v2.0]

JSBSim startup beginning ...

  Output data set: 0  
Reading Aircraft Configuration File: rascal
                            Version: 2.0

This aircraft model is a BETA release!!!

This aircraft model probably will not fly as expected.

Use this model for development purposes ONLY!!!

  Description:   Models a rascal
  Model Author:  Author Name
  Creation Date: Creation Date
  Version:       Version

  Aircraft Metrics:
    WingArea: 10.57
    WingSpan: 9.17
    Incidence: 0
    Chord: 1.15
    H. Tail Area: 1.69
    H. Tail Arm: 3.28
    V. Tail Area: 1.06
    V. Tail Arm: 0
    Eyepoint (x, y, z): 20 , 0 , 5
    Ref Pt (x, y, z): 37.4 , 0 , 0
    Visual Ref Pt (x, y, z): 0 , 0 , 0

  Mass and Balance:
    baseIxx: 1.95 slug-ft2
    baseIyy: 1.55 slug-ft2
    baseIzz: 1.91 slug-ft2
    baseIxy: -0 slug-ft2
    baseIxz: 0 slug-ft2
    baseIyz: -0 slug-ft2
    Empty Weight: 13 lbm
    CG (x, y, z): 36.4 , 0 , 4

[... 352 lines cut ...]

### To Reproduce

Follow docs for running JSBSim SITL: https://docs.px4.io/main/en/sim_jsbsim/index.html

### Expected behavior

Should be able to connect from QGCS, arm and command takeoff via GCS or commandline:

```
commander arm
commander takeoff
```

### Flight Log

https://logs.px4.io/plot_app?log=76f960d7-821d-44c0-8b66-3857b6ed16d6

### Software Version

main

### Vehicle type

None

Jaeyoung-Lim, 2025-03-18:
JSBSim has not been maintained for a while.

This is what I get on Ubuntu Focal JSBsim v1.2.1
```
ERROR [simulator_mavlink] Number of simulated accelerometer 80 out of range. Max: 3

```

Jaeyoung-Lim, 2025-03-18:
@simfinite Another issue that might be relevant: https://github.com/JSBSim-Team/jsbsim/issues/325

simfinite, 2025-03-19:
> [@simfinite](https://github.com/simfinite) Another issue that might be relevant: [JSBSim-Team/jsbsim#325](https://github.com/JSBSim-Team/jsbsim/issues/325)

Interesting, I haven't run into this issue. The FG visualization seems to work on my end with the following versions:
```
FlightGear version: 2020.3.19
JSBSim Version: 1.2.1
```

I installed FG via saiarcot895 ppa and JSBSim from deb package on [github release page](https://github.com/JSBSim-Team/jsbsim/releases/tag/v1.2.1).

I did just find out that I wasn't running the JSBSim version, I intended to run. Apparently, the `JSBSim-devel` package does not include the binaries, so the non-devel `JSBSim` package must be installed as well, otherwise SITL will pick up the JSBSim binary installed with FG. This should be changed in the [documentation](https://docs.px4.io/main/en/sim_jsbsim/index.html#installation-ubuntu-linux) along with updated version numbers once we have a running setup.

simfinite, 2025-03-19:
Yes, both FG and QGCS show Zurich airport. Had to find the `download scenery` switch in the FG settings.

Btw, I am on currently on Jammy and have also been testing on Noble. Not sure if the above mentioned versions are available for Focal, as used in https://github.com/JSBSim-Team/jsbsim/issues/325

closed 2026-01-16 by github-actions[bot]