PX4/PX4-Autopilot #24477: [Bug] fastdds return error "Failed to checkout tag: '2.12.x'" when follow the px4 office doc

yeshangjun, 2025-03-09:
### Describe the bug

I try to install the ros,  Micro XRCE-DDS Agent & Client, following the instuction in 
https://docs.px4.io/main/en/ros2/user_guide.html#humble

**platform: ubuntu 22.04**
**px4: v1.15.2-1beta**
when I try the command in 
#make
```
git clone -b v2.4.2 https://github.com/eProsima/Micro-XRCE-DDS-Agent.git
cd Micro-XRCE-DDS-Agent
mkdir build
cd build
cmake ..
make
sudo make install
sudo ldconfig /usr/local/lib/
```

returns error:
[ 35%] Performing build step for 'fastcdr'
Consolidate compiler generated dependencies of target fastcdr
[100%] Built target fastcdr
[ 38%] Performing install step for 'fastcdr'
[100%] Built target fastcdr
Install the project...
-- Install configuration: "Release"
-- Up-to-date: /home/athena/Micro-XRCE-DDS-Agent/build/temp_install/fastcdr-1.1.1/include/fastcdr
-- Up-to-date: /home/athena/Micro-XRCE-DDS-Agent/build/temp_install/fastcdr-1.1.1/include/fastcdr/fastcdr_dll.h
-- Up-to-date: /home/athena/Micro-XRCE-DDS-Agent/build/temp_install/fastcdr-1.1.1/include/fastcdr/FastCdr.h
-- Up-to-date: /home/athena/Micro-XRCE-DDS-Agent/build/temp_install/fastcdr-1.1.1/include/fastcdr/FastBuffer.h
-- Up-to-date: /home/athena/Micro-XRCE-DDS-Agent/build/temp_install/fastcdr-1.1.1/include/fastcdr/eProsima_auto_link.h
-- Up-to-date: /home/athena/Micro-XRCE-DDS-Agent/build/temp_install/fastcdr-1.1.1/include/fastcdr/Cdr.h
-- Up-to-date: /home/athena/Micro-XRCE-DDS-Agent/build/temp_install/fastcdr-1.1.1/include/fastcdr/exceptions
-- Up-to-date: /home/athena/Micro-XRCE-DDS-Agent/build/temp_install/fastcdr-1.1.1/include/fastcdr/exceptions/BadParamException.h
-- Up-to-date: /home/athena/Micro-XRCE-DDS-Agent/build/temp_install/fastcdr-1.1.1/include/fastcdr/exceptions/Exception.h
-- Up-to-date: /home/athena/Micro-XRCE-DDS-Agent/build/temp_install/fastcdr-1.1.1/include/fastcdr/exceptions/NotEnoughMemoryException.h
-- Up-to-date: /home/athena/Micro-XRCE-DDS-Agent/build/temp_install/fastcdr-1.1.1/include/fastcdr/config.h
-- Up-to-date: /home/athena/Micro-XRCE-DDS-Agent/build/temp_install/fastcdr-1.1.1/lib/libfastcdr.so.1.1.1
-- Up-to-date: /home/athena/Micro-XRCE-DDS-Agent/build/temp_install/fastcdr-1.1.1/lib/libfastcdr.so
-- Up-to-date: /home/athena/Micro-XRCE-DDS-Agent/build/temp_install/fastcdr-1.1.1/lib/cmake/fastcdr/fastcdr-shared-targets.cmake
-- Up-to-date: /home/athena/Micro-XRCE-DDS-Agent/build/temp_install/fastcdr-1.1.1/lib/cmake/fastcdr/fastcdr-shared-targets-release.cmake
-- Up-to-date: /home/athena/Micro-XRCE-DDS-Agent/build/temp_install/fastcdr-1.1.1/lib/cmake/fastcdr/fastcdr-config.cmake
-- Up-to-date: /home/athena/Micro-XRCE-DDS-Agent/build/temp_install/fastcdr-1.1.1/lib/cmake/fastcdr/fastcdr-config-version.cmake
-- Up-to-date: /home/athena/Micro-XRCE-DDS-Agent/build/temp_install/fastcdr-1.1.1/share/fastcdr/LICENSE
[ 41%] Performing test step for 'fastcdr'
[ 44%] Completed 'fastcdr'
[ 50%] Built target fastcdr
[ 52%] Creating directories for 'fastdds'
[ 55%] Performing download step (git clone) for 'fastdds'
**Cloning into 'fastdds'...**
**fatal: invalid reference: 2.12.x**
**CMake Error at /home/athena/Micro-XRCE-DDS-Agent/build/fastdds/tmp/fastdds-gitclone.cmake:40 (message):**
Failed to checkout tag: '2.12.x' 

make[2]: *** [CMakeFiles/fastdds.dir/build.make:101: fastdds/src/fastdds-stamp/fastdds-download] Error 1
make[1]: *** [CMakeFiles/Makefile2:141: CMakeFiles/fastdds.dir/all] Error 2
make: *** [Makefile:91: all] Error 2

### Vehicle type

None

yeshangjun, 2025-03-11:
> I had the same problem and I found the solution [here](https://github.com/eProsima/Micro-XRCE-DDS-Agent/issues/370)
> 
> You need to modify the [line 99 of CMakeLists.txt](https://github.com/eProsima/Micro-XRCE-DDS-Agent/blob/57d086216d01ec43121845d385894a25987f8a2c/CMakeLists.txt#L99) so: `set(_fastdds_tag 2.12.1)`

Thx, problem solved.

federicociresola, 2025-03-11:
perfect, can @yeshangjun close the issue?

closed 2025-03-11 by yeshangjun

bentwookie, 2025-04-14:
What's completed about this? Is my dockerfile supposed to sed this change every time? Should the user guide tell people to make this change?

yblee1223, 2025-05-07:
I had a same issue, because of the version of fast DDS, So I changed it to         
```
set(_fastdds_version 2.13)
set(_fastdds_tag v2.13.6)
```
to the acceptable version 
and It worked

harunkurtdev, 2025-05-22:
Hello,

I wrote a Dockerfile for this issue 

Would you view from bellow url

https://gist.github.com/harunkurtdev/7e20d3e4372dc26402d7c64ee3345dfb

yocute, 2025-05-26:
@aaronxavier hey you saved my life. Thank you so much!

amirrezasadeqi, 2025-06-30:
> I had a same issue, because of the version of fast DDS, So I changed it to
> 
> ```
> set(_fastdds_version 2.13)
> set(_fastdds_tag v2.13.6)
> ```
> 
> to the acceptable version and It worked

I think the problem actually is that the `v` is omitted from the beginning of the tag.

milesfrain, 2025-07-15:
A better solution is if the eProsima folks restore the deleted `2.12.x` branch.

Until that happens, this should still be considered an open issue. Anyone else following the PX4 ROS2 guide will also encounter this error, and have to rummage for this workaround.

Full explanation here:
https://github.com/eProsima/Micro-XRCE-DDS-Agent/issues/370#issuecomment-3074205167

tuananh0606-tech, 2025-09-04:
> I had the same problem and I found the solution [here](https://github.com/eProsima/Micro-XRCE-DDS-Agent/issues/370)
> You need to modify the [line 99 of [...]

I have to change ```set(_fastdds_version 2.13)
set(_fastdds_tag v2.13.x)''
, and It worked very well