PX4/PX4-Autopilot #24487: Test GCC 13 builds from new PX4-Dev container

mrpollo, 2025-03-11:
### Describe the bug

We need to make sure the build artifacts with an updated container using GCC 13. See #24022 for context

- Pixhawk 4 - Download [px4_fmu-v6c_default.zip](https://github.com/user-attachments/files/19168706/px4_fmu-v6c_default.zip)
- Pixhawk 6C - Download [px4_fmu-v4_default.zip](https://github.com/user-attachments/files/19168708/px4_fmu-v4_default.zip)
- NXP VMU-RT1176 - Download [px4_fmu-v6xrt_default.zip](https://github.com/user-attachments/files/19168718/px4_fmu-v6xrt_default.zip)

### Vehicle type

None

mrpollo, 2025-03-11:
@PX4/testflights please post logs here!

DroneWuKong, 2025-03-20:
For the Pixhawk 4 firmware, any chance I could get it for the Pixhawk 4 FMUv5? 

It's been a crazy couple weeks, Getting back to this to get you some data now. Will have logs posted for Pixhawk 6C today.

mrpollo, 2025-03-20:
sorry i uploaded the wrong firmware, but I have corrected the mistake and updated the link above to match fmuv5, here's a link just in case [px4_fmu-v5_default.zip](https://github.com/user-attachments/files/19375382/px4_fmu-v5_default.zip)

mrpollo, 2025-05-08:
Closed by #24022 

Logs from test flights can be found at the links below. All tests passed.
* [Pixhawk 6C](https://github.com/PX4/PX4-Autopilot/pull/24022#issuecomment-2748203489)
* [NXP VMU-RT1176](https://github.com/PX4/PX4-Autopilot/pull/24022#issuecomment-2751282121)
* [Pixhawk 4](https://github.com/PX4/PX4-Autopilot/pull/24022#issuecomment-2751765826)

closed 2025-05-08 by mrpollo