PX4/PX4-Autopilot #24496: [Bug] takeoff altitude is 1m short

dakejahl, 2025-03-12:
### Describe the bug

The CI failures are real
https://github.com/PX4/PX4-Autopilot/actions/runs/13820943813/job/38665877341?pr=24495
```
[  41.836|mavsdk_tests] Figure eight execution clockwise
[  41.836|mavsdk_tests] -------------------------------------------------------------------------------
[  41.836|mavsdk_tests] ../../../test/mavsdk_tests/test_vtol_figure_eight.cpp:39
[  41.836|mavsdk_tests] ...............................................................................
[  41.836|mavsdk_tests] ../../../test/mavsdk_tests/autopilot_tester.cpp:225: FAILED:
```
which corresponds to `AutopilotTester::wait_until_altitude`, and in the test it is set to `const float takeoff_altitude = 20.f;`

It looks like the takeoff altitude is set to 20m but the vehicle only climbs to 19m
https://logs.px4.io/plot_app?log=d3a7ff11-2703-4a7a-9942-dee227747676

### Vehicle type

None

dakejahl, 2025-03-12:
@bresch can you take a look?

dakejahl, 2025-03-12:
@MaEtUgR can you take a look too?

dakejahl, 2025-03-13:
This broke recently... this PR used to pass checks but is now failing on the new test (takeoff and hold altitude test...)
https://github.com/PX4/PX4-Autopilot/pull/24396
https://logs.px4.io/plot_app?log=ce217a84-f73c-47c6-afba-e8b40430d237

Need to get this test in and disallow overriding CI so we stop breaking things..

dakejahl, 2025-03-13:
Ahh I think it was this PR that merged yesterday
https://github.com/PX4/PX4-Autopilot/pull/24454
@sfuhrer can you take a look?

sfuhrer, 2025-03-13:
Thanks for raising it @dakejahl ! For once the CI was helpful, just hard to see the real failures between all the false positive ones.

dakejahl, 2025-03-13:
Yup hopefully we can get the CI fixed up for good so we can start relying on it again

closed 2025-03-17 by dakejahl