PX4/PX4-Autopilot #24497: [Bug] [Docs] Add docs for gz plugins

hamishwillee, 2025-03-13:
### Describe the bug

Add docs for #24441 - (gz: use server config file for loading world plugins) as per https://github.com/PX4/PX4-Autopilot/pull/24441#issuecomment-2695945337 and https://github.com/PX4/PX4-user_guide/pull/3628/files#r1990392300

@dakejahl and @hamishwillee to talk

### Vehicle type

None

hamishwillee, 2025-03-13:
@dakejahl Note also https://github.com/PX4/PX4-user_guide/issues/3641#issuecomment-2720541759

  > I was focusing on: https://docs.px4.io/main/en/sim_gazebo_gz/index.html
In particular in the section "Adding New Worlds and Models", subsection "Adding a World". There is a reference to a `CMakeLists.txt` file that should be modified to generate the targets. However, checking the git history I see that such list has been removed in commit [6dc39d9deb6f1de4c1bb97461193aba8ff679c59](https://github.com/PX4/PX4-Autopilot/commit/6dc39d9deb6f1de4c1bb97461193aba8ff679c59). Is it possible that the documentation is now outdated?

Is this something that can be fixed quickly or does it have to wait on this discussion?

hamishwillee, 2025-04-10:
@dakejahl I'm off for two weeks. Are you at the point where this can be documented? If so, feel free to kick it off.

dakejahl, 2025-04-10:
Yeah, just need to find the time to do it. I also want to take one more pass at my optical flow plugin because I realized I did something wrong and want to fix it first.

hamishwillee, 2025-04-30:
I'm back. This is your nag!

closed 2025-05-28 by hamishwillee