PX4/PX4-Autopilot #24505: Drone starts crashing after Auto-Tune (PX4, OctaQuad, Larger Inertia, Pixhawk 6X)

MuhammadBilal1, 2025-03-13:
### Describe the bug

## Drone Setup:

- Quadcopter Type: OctaQuad (Coaxial Octocopter)
- Flight Controller: PX4 1.15.2
- GPS: Integrated
- Flight Controller Hardware: Pixhawk 6X
- Airframe: Generic 10" Octocopter Coaxial Geometry
- Body Structure: Multilayered Carbon Fiber with removable arms
- Length (Nose to Tail): 230 cm
- Width (Right Motor to Left Motor): 352 cm

### I am attempting to tune the PID settings for my quadcopter, starting with manual tuning to prepare it for the Auto-Tune process on the Rate Controller.

Once I reached a point where I felt ready to start the Auto-Tune, I initiated the Auto-Tune flight, and the parameters were successfully saved. However, immediately after the Auto-Tune flight, the drone flipped shortly after takeoff.

After repairing the structure (due to the crash), I reverted the parameters back to the Auto-Tune settings and attempted manual tuning again. Unfortunately, I was unable to achieve optimal PID tuning. I'm mainly experiencing wobbling on the pitch axis.

### What I need?
Please help me identify the area of problem and how I can fix that?

### Expected behavior

The flight should be stable and not wobble. Currently, the drone is wobbling on the pitch or roll axis and began flipping after the autotuned flight.

### Flight Log

#### Pre Tune Flight

https://review.px4.io/plot_app?log=5b1df7f2-8924-4acc-ab99-a3ae9a517ae6

=========================================================================
#### Pre Tune Flight

https://review.px4.io/plot_app?log=e4ebbd5e-e8fe-4fe4-beb9-574d6c0a8883

=========================================================================
## Auto Tune

https://review.px4.io/plot_app?log=94bcd5a5-0a12-4e96-9c1b-c7e53612c9db

=========================================================================
## Post Autotune (Crash on Takeoff)

https://review.px4.io/plot_app?log=4c63fd8c-40ac-451c-a2c5-2f438297a5d3

=========================================================================
#### Manual Pitch Tune

MC_PITCHRATE_P  > 0.2
MC_PITCHRATE_I  > 0.3
MC_PITCHRATE_D > 0.003

https://review.px4.io/plot_app?log=65e27969-8c72-4ad8-a154-94fbe738b2e8

========================================================================
#### Manual Pitch Tune

MC_PITCHRATE_P > 0.18
MC_PITCHRATE_I > 0.25
MC_PITCHRATE_D > 0.0028

https://review.px4.io/plot_app?log=f5e77b52-3c45-4149-bb06-3d55a15aeeb8

=========================================================================
#### Manual Pitch Tune

MC_PITCHRATE_P > 0.15
MC_PITCHRATE_I > 0.21

https://review.px4.io/plot_app?log=20d60ffb-4f0b-46ca-bb6d-21a0a35f6fc0

=========================================================================
## Manual Pitch Tune (Crash on Takeoff)

MC_PITCHRATE_P  > 0.2
MC_PITCHRATE_I  > 0.35
MC_PITCHRATE_D > 0.004

https://review.px4.io/plot_app?log=f2767777-c7fe-499e-b73c-8eff56c432d4

### Software Version

PX4 1.15.2

### Flight controller

Pixhawk 6X

### Vehicle type

Multicopter

crazy-robotics, 2025-03-14:
checkout the following [discussion](https://discuss.px4.io/t/very-large-uav-drone-stability-issues/44324)

MuhammadBilal1, 2025-03-20:
@dakejahl 
Can you please check my issue and suggest something?
Thank you

bresch, 2025-03-24:
@MuhammadBilal1 Same problem as in https://discuss.px4.io/t/very-large-uav-drone-stability-issues/44324/4 , you need to fix the yaw effectiveness problem first. Also, this is not a firmware bug, please re-open an issue in [https://discuss.px4.io](https://discuss.px4.io) if needed

closed 2025-03-24 by bresch