PX4/PX4-Autopilot #24511: [Bug] Terrain Following and Collision Prevention doesn't work simultaneously

mars2860, 2025-03-14:
### Describe the bug

Terrain Following works when we set MPC_ALT_MODE=Terrain Following and MPC_POS_MODE=Direct Velocity.
Collision Prevention works when we set CP_DIST > 0 and MPC_POS_MODE=Acceleration based (since Commit 0cd6a55). Collision prevention stops to work if I set MPC_POS_MODE=Direct Velocity. And Terrain Following stops to work if I set MPC_POS_MODE=Acceleration Based. Thus these two modes can't work simultaneously.

### To Reproduce

`make px4_sitl gz_x500_lidar_down`

Set MPC_ALT_MODE=Terrain Following and MPC_POS_MODE=Acceleration based. Terrain following will not work.

`make px4_sitl gz_x500_lidar_2d_walls`

Set CP_DIST=3 and MPC_POS_MODE=Direct Velocity. Collision Prevention will not work.

### Vehicle type

None

bresch, 2025-03-19:
> the bug is here: this line set altitude setpoint as NAN when terrain following works

Sorry but I don't get it, when terrain following is active, `_terrain_hold` is true, so it won't enter this condition, right?

mars2860, 2025-03-20:
No, when terrain following is active `_terrain_hold` is false. This flag is true only when a vehicle stops. You can see it in the code below

```
void FlightTaskManualAltitude::_updateAltitudeLock() {
...

bool not_moving = spd_xy < 0.5f * _param_mpc_hold_max_xy.get();

			if (!stick_input && not_moving && PX4_ISFINITE(_dist_to_bottom)) {
				// Start using distance to ground
				_terrain_hold = true;

...
}
```

mars2860, 2025-11-19:
I have fixed this on my local copy of PX4 as I have mentioned above and it works

closed 2026-05-04 by julianoes