PX4/PX4-Autopilot #24515: [Bug] DShot Serial Telemetry without Bidirectional Enabled is Broken

AlexKlimaj, 2025-03-14:
### Describe the bug

With the new Bidirectional Dshot PR that just went in, serial telemetry is broken when bidirectional dshot is not enabled.

### Vehicle type

None

AlexKlimaj, 2025-03-19:
It looks like the telemetry is not being requested?

[image]

AlexKlimaj, 2025-03-20:
RPM data on bidirectional with serial telem. Need to check without serial telem enabled

https://review.px4.io/plot_app?log=f4a34197-7a72-47b0-b5df-c144191bc468

[image]
[image]

Without serial telem enabled.

https://review.px4.io/plot_app?log=b042fdb4-1fcf-484d-bbb3-6aa97b7d284e

[image]

AlexKlimaj, 2025-03-20:
@julianoes

dakejahl, 2025-03-20:
I'll work on this on Monday. I think we should probably just round-robin all of the timer channels on a single DMA. This is what Ardupilot and Betaflight do, we should be okay with 1/4th rate feedback.

closed 2025-03-27 by AlexKlimaj