PX4/PX4-Autopilot #24594: [Bug] No valid position estimate

crazy-robotics, 2025-03-24:
### Describe the bug

I am spawning following [xacro file](https://github.com/crazy-robotics/dronedoc/blob/master/dronedoc/models/iris_depth_camera/xacro_sitl/urdf/iris_base.xacro) where gps is called  
```
  <xacro:gps_plugin_macro
    namespace="${namespace}"
    gps_noise="true"
    >
  </xacro:gps_plugin_macro>
```
and plugin I can see in gazebo 
[image]
but on QGC following messages appears 
```
Navigation error: No valid position estimate
Navigation error: No valid global position estimate
```
[image]

### To Reproduce

git clone the following [repo](https://github.com/crazy-robotics/dronedoc/tree/master) and 
```
roslaunch dronedoc iris_moveit_1.launch 
```

### Expected behavior

```
Navigation error: No valid position estimate
Navigation error: No valid global position estimate
```

### Vehicle type

None

crazy-robotics, 2025-03-24:
I have source px4 in .bash rc 
```
cd PX4-Autopilot
source Tools/simulation/gazebo-classic/setup_gazebo.bash $(pwd) $(pwd)/build/px4_sitl_default
export ROS_PACKAGE_PATH=$ROS_PACKAGE_PATH:$(pwd)
export ROS_PACKAGE_PATH=$ROS_PACKAGE_PATH:$(pwd)/Tools/simulation/gazebo-classic/sitl_gazebo-classic
```
and following is the output 
```
mubashir@mubashir:~$ source .bashrc 
GAZEBO_PLUGIN_PATH :/home/mubashir/PX4-Autopilot/build/px4_sitl_default/build_gazebo-classic:/home/mubashir/PX4-Autopilot/build/px4_sitl_default/build_gazebo-classic:/home/mubashir/PX4-Autopilot/build/px4_sitl_default/build_gazebo-classic
GAZEBO_MODEL_PATH :/home/mubashir/PX4-Autopilot/Tools/simulation/gazebo-classic/sitl_gazebo-classic/models:/home/mubashir/PX4-Autopilot/Tools/simulation/gazebo-classic/sitl_gazebo-classic/models:/home/mubashir/PX4-Autopilot/Tools/simulation/gazebo-classic/sitl_gazebo-classic/models
LD_LIBRARY_PATH /home/mubashir/moveit_iris/devel/lib:/home/mubashir/catkin_ws/devel/lib:/opt/ros/noetic/lib:/opt/ros/noetic/lib/x86_64-linux-gnu:/home/mubashir/PX4-Autopilot/build/px4_sitl_default/build_gazebo-classic:/home/mubashir/PX4-Autopilot/build/px4_sitl_default/build_gazebo-classic:/home/mubashir/PX4-Autopilot/build/px4_sitl_default/build_gazebo-classic

```

bresch, 2025-03-24:
What do you get if you type the following commands in the mavlink shell?
- `listener vehicle_gps_position`
- `listener estimator_status_flags`
- `listener estimator_gps_status`
- `listener vehicle_local_position`

Jaeyoung-Lim, 2025-03-24:
This is an issue on the model generated from xacro in the repository you are using, not an issue related to px4.

I suggest you post the issue in dronedoc directly

closed 2025-03-24 by Jaeyoung-Lim

crazy-robotics, 2025-03-24:
here is the ouput of mavlink console 
1.
 ```
listener vehicle_gps_position

TOPIC: vehicle_gps_position
```
2)

```
listener estimator_status_flags

TOPIC: estimator_status_flags 3 instances

Instance 0:
 estimator_status_flags
    timestamp: 124012000 (0.624000 seconds ago)
    timestamp_sample: 123860000 (152000 us before timestamp)
    control_status_changes: 7
    fault_status_changes: 0
    innovation_fault_status_changes: 2
    cs_tilt_align: True
    cs_yaw_align: True
    cs_gps: False
    cs_opt_flow: False
    cs_mag_hdg: True
    cs_mag_3d: False
    cs_mag_dec: True
    cs_in_air: False
    cs_wind: False
    cs_baro_hgt: True
    cs_rng_hgt: False
    cs_gps_hgt: False
    cs_ev_pos: False
    cs_ev_yaw: False
    cs_ev_hgt: False
    cs_fuse_beta: False
    cs_mag_field_disturbed: False
    cs_fixed_wing: False
    cs_mag_fault: False
    cs_fuse_aspd: False
    cs_gnd_effect: False
    cs_rng_stuck: False
    cs_gnss_yaw: False
    cs_mag_aligned_in_flight: False
    cs_ev_vel: False
    cs_synthetic_mag_z: False
    cs_vehicle_at_rest: True
    cs_gnss_yaw_fault: False
    cs_rng_fault: False
    cs_inertial_dead_reckoning: True
    cs_wind_dead_reckoning: False
    cs_rng_kin_consistent: False
    cs_fake_pos: False
    cs_fake_hgt: False
    cs_gravity_vector: True
    cs_mag: True
    cs_ev_yaw_fault: False
    cs_mag_heading_consistent: True
    cs_aux_gpos: False
    cs_rng_terrain: False
    cs_opt_flow_terrain: False
    cs_valid_fake_pos: True
    cs_constant_pos: False
    cs_baro_fault: False
    fs_bad_mag_x: False
    fs_bad_mag_y: False
    fs_bad_mag_z: False
    fs_bad_hdg: False
    fs_bad_mag_decl: False
    fs_bad_airspeed: False
    fs_bad_sideslip: False
    fs_bad_optflow_x: False
    fs_bad_optflow_y: False
    fs_bad_acc_vertical: False
    fs_bad_acc_clipping: False
    reject_hor_vel: False
    reject_ver_vel: False
    reject_hor_pos: False
    reject_ver_pos: False
    reject_yaw: False
    reject_airspeed: False
    reject_sideslip: False
    reject_hagl: False
    reject_optflow_x: False
    reject_optflow_y: False

Instance 1:

Instance 2:
pxh> 
```
3. 
```
listener estimator_gps_status

TOPIC: estimator_gps_status 3 instances

Instance 0:

Instance 1:

Instance 2:
pxh> 
```
4.
```
listener vehicle_local_position

TOPIC: vehicle_local_position
 vehicle_local_position
    timestamp: 173704000 (0.000000 seconds ago)
    timestamp_sample: 173704000 (0 us before timestamp)
    ref_timestamp: 0
    ref_lat: nan
    ref_lon: nan
    x: -1.02372
    y: -0.03582
    z: 0.12072
    delta_xy: [0.00000, 0.00000]
    delta_z: 0.00000
    vx: 0.00838
    vy: -0.00176
    vz: 0.03210
    z_deriv: -0.00018
    delta_vxy: [1.15132, 0.07089]
    delta_vz: 0.00535
    ax: -0.01178
    ay: -0.04972
    az: 0.02559
    heading: 1.70155
    heading_var: 0.01173
    unaided_heading: 0.01530
    delta_heading: 1.74090
    tilt_var: 0.00003
    ref_alt: nan
    dist_bottom: 0.10000
    dist_bottom_var: 0.01000
    delta_dist_bottom: 0.00000
    eph: 0.01160
    epv: 0.28352
    evh: 0.03454
    evv: 0.05278
    vxy_max: inf
    vz_max: inf
    hagl_min: inf
    hagl_max_z: inf
    hagl_max_xy: inf
    xy_valid: False
    z_valid: True
    v_xy_valid: False
    v_z_valid: True
    xy_reset_counter: 1
    z_reset_counter: 1
    vxy_reset_counter: 1
    vz_reset_counter: 1
    heading_reset_counter: 1
    heading_good_for_control: False
    xy_global: False
    z_global: False
    dist_bottom_valid: False
    dist_bottom_reset_counter: 1
    dist_bottom_sensor_bitfield: 0
    dead_reckoning: True
pxh> 
```

crazy-robotics, 2025-03-27:
The error has solved Check the following comment https://github.com/PX4/PX4-SITL_gazebo-classic/issues/660#issuecomment-2757012214