PX4/PX4-Autopilot #24595: [Bug] Gazebo floods network with multicast packets depending on SITL launch method

MaxwellsEquation, 2025-03-24:
### Describe the bug

Gazebo multicast packets will be published en masse to the whole network instead of just being published to the local host.

### To Reproduce

On Arch Linux, linux-zen kernel, with PX4 running inside a distrobox ubuntu container.

In particular:
Launching using make px4_sitl_gzx500 leads to packets remaining local:
sudo tcpdump -i any udp:
22:27:49.959357 lo    In  IP localhost.10317 > 239.255.0.7.10317: UDP, length 224
22:27:49.962896 lo    In  IP localhost.18570 > localhost.14550: UDP, length 40
22:27:49.962906 lo    In  IP localhost.14580 > localhost.14540: UDP, length 40
22:27:49.963209 lo    In  IP localhost.10317 > 239.255.0.7.10317: UDP, length 86
22:27:49.963272 lo    In  IP localhost.10317 > 239.255.0.7.10317: UDP, length 221
22:27:49.963276 lo    In  IP localhost.10317 > 239.255.0.7.10317: UDP, length 230
22:27:49.963280 lo    In  IP localhost.10317 > 239.255.0.7.10317: UDP, length 221
22:27:49.963284 lo    In  IP localhost.10317 > 239.255.0.7.10317: UDP, length 248
^C

Launching with parameters breaks everything:
PX4_SYS_AUTOSTART=4001 PX4_GZ_MODEL_POSE="0,1" PX4_SIM_MODEL=gz_x500 ./build/px4_sitl_default/bin/px4 -i 1
sudo tcpdump -i any udp:
22:29:20.828044 wlp9s0 Out IP vk-leenux.10317 > 239.255.0.7.10317: UDP, length 222
22:29:20.828052 wlp9s0 Out IP vk-leenux.10317 > 239.255.0.7.10317: UDP, length 222
22:29:20.828058 wlp9s0 Out IP vk-leenux.10317 > 239.255.0.7.10317: UDP, length 222
22:29:20.828065 wlp9s0 Out IP vk-leenux.10317 > 239.255.0.7.10317: UDP, length 210
22:29:20.828072 wlp9s0 Out IP vk-leenux.10317 > 239.255.0.7.10317: UDP, length 226
22:29:20.828077 wlp9s0 Out IP vk-leenux.10317 > 239.255.0.7.10317: UDP, length 224
22:29:20.828084 wlp9s0 Out IP vk-leenux.10317 > 239.255.0.7.10317: UDP, length 224
22:29:20.828151 wlp9s0 Out IP vk-leenux.10317 > 239.255.0.7.10317: UDP, length 224
22:29:20.828318 wlp9s0 Out IP vk-leenux.10317 > 239.255.0.7.10317: UDP, length 222
22:29:20.828349 wlp9s0 Out IP vk-leenux.10317 > 239.255.0.7.10317: UDP, length 222

Note that this also leads to instability in the Gazebo simulator (very low frame rate) and a myriad of error messages:
pxh> ERROR [sensors] Accel #0 fail:  TIMEOUT!	
ERROR [health_and_arming_checks] angular velocity no longer valid (timeout)
ERROR [vehicle_magnetometer] MAG #0 failed:  TIMEOUT!	
WARN  [health_and_arming_checks] Preflight Fail: No valid data from Accel 0
WARN  [health_and_arming_checks] Preflight Fail: No valid data from Baro 0
WARN  [health_and_arming_checks] Preflight Fail: No valid data from Gyro 0
WARN  [health_and_arming_checks] Preflight Fail: No valid data from Compass 0
WARN  [health_and_arming_checks] Preflight Fail: Battery unhealthy
WARN  [health_and_arming_checks] Preflight Fail: No connection to the ground control station	
ERROR [vehicle_air_data] BARO #0 failed:  TIMEOUT!	
WARN  [health_and_arming_checks] Preflight Fail: No connection to the ground control station	
ERROR [vehicle_angular_velocity] unable to find or subscribe to selected sensor (1310988)
ERROR [health_and_arming_checks] angular velocity no longer valid (timeout)
WARN  [health_and_arming_checks] Preflight Fail: No valid data from Accel 0
WARN  [health_and_arming_checks] Preflight Fail: No valid data from Baro 0
WARN  [health_and_arming_checks] Preflight Fail: No valid data from Gyro 0
WARN  [health_and_arming_checks] Preflight Fail: No valid data from Compass 0
WARN  [health_and_arming_checks] Preflight Fail: No connection to the ground control station	
ERROR [vehicle_air_data] BARO #0 failed:  TIMEOUT!	
WARN  [health_and_arming_checks] Preflight Fail: No connection to the ground control station	
pxh> ERROR [sensors] Accel #0 fail:  TIMEOUT!
ERROR [health_and_arming_checks] angular velocity no longer valid (timeout)
WARN  [health_and_arming_checks] Preflight Fail: No valid data from Accel 0
WARN  [health_and_arming_checks] Preflight Fail: No valid data from Baro 0
WARN  [health_and_arming_checks] Preflight Fail: No valid data from Gyro 0
WARN  [health_and_arming_checks] Preflight Fail: No valid data from Compass 0
WARN  [health_and_arming_checks] Preflight Fail: No connection to the ground control station	
ERROR [vehicle_air_data] BARO #0 failed:  TIMEOUT!	
WARN  [mavlink] Event dropped (52, 68)
WARN  [mavlink] Event dropped (52, 68)
WARN  [mavlink] Event dropped (52, 68)
WARN  [mavlink] Event dropped (52, 68)

### Expected behavior

The output of sudo tcpdump -i any udp should be the same, the behavior of the simulator should be the same.

### Software Version

1.16--most up to date commit as of posting of the report

### Flight controller

simulation

### Vehicle type

None

### Additional context

The issue is resolved by calling PX4_SYS_AUTOSTART=4001 GZ_TRANSPORT_LOCALHOST_ONLY=1 IGN_TRANSPORT_DISABLE_MULTICAST=1 IGN_IP=127.0.0.1 GZ_IP=127.0.0.1 PX4_SIM_MODEL=gz_x500 ./build/px4_sitl_default/bin/px4 -i 1 which fixes multicast and makes multi-drone sim behave as expected. I think this is because if you initialize the sim this way, gz_env.sh might not be executed?

dakejahl, 2025-03-24:
I think this is fixed in https://github.com/PX4/PX4-Autopilot/pull/24526/files

Can you test and report back?

MaxwellsEquation, 2025-04-03:
It's fixed!

closed 2025-04-03 by MaxwellsEquation