PX4/PX4-Autopilot #24598: [Bug] px4-rc.simulator set gc_command in wrong place

yexuanyang, 2025-03-24:
### Describe the bug

I just want to boot px4 by `PX4_GZ_STANDALONE=1 make px4_sitl gz_x500`, and I run the script in https://github.com/PX4/PX4-gazebo-models/blob/main/simulation-gazebo to start the model, and in theory, px4 will connect to the gazebo and stop waiting for gazebo world. But it won't, and reach the limit retry times, then exit.

### To Reproduce

1. Just build px4 normally, `make px4_sitl gz_x500`
2. Install the script in https://github.com/PX4/PX4-gazebo-models/blob/main/simulation-gazebo
3. Uncomment the line `set +x` in rcS to see the debug info.
4. Run `PX4_GZ_STANDALONE=1 make px4_sitl gz_x500` in a terminal, then run script installed at step 2 in a different terminal.
5. Well, after 30 retries, px4 will exit. You can see `gz_command` is set to NULL because it is only set in the branch of `if [ -z "${PX4_GZ_STANDALONE}" ]; then`.

### Expected behavior

px4 should connect to the gazebo simulator after the script [simulation-gazebo](https://github.com/PX4/PX4-gazebo-models/blob/main/simulation-gazebo) run.

### Screenshot / Media

[image]

### Flight Log

No log here

### Software Version

checkout at commit e0d15b7a80. QGC version v4.4.3，`simulation-gazebo` script checkout commit c3905a7, url https://github.com/PX4/PX4-gazebo-models/blob/main/simulation-gazebo

$ gz sim --version
Gazebo Sim, version 8.9.0
Copyright (C) 2018 Open Source Robotics Foundation.
Released under the Apache 2.0 License.

### Vehicle type

None

yexuanyang, 2025-03-24:
It seems that in the latest [px4-rc.simulator script](https://github.com/PX4/PX4-Autopilot/blob/main/ROMFS/px4fmu_common/init.d-posix/px4-rc.simulator), function `check_scene_info` is placed in the `if [ -z "${PX4_GZ_STANDALONE}" ];` branch, and the gz_command used is fine. But the gz_command here is still empty 
https://github.com/PX4/PX4-Autopilot/blob/ca2ed655b8cbf947e853ac5000cbd543edd74d7d/ROMFS/px4fmu_common/init.d-posix/px4-rc.simulator#L157-L163

dakejahl, 2025-03-24:
@yexuanyang it should be fixed in https://github.com/PX4/PX4-Autopilot/pull/24600

Please try and and let me know

yexuanyang, 2025-03-25:
Good! It is fine now.

PR #24600 (2025-03-25): sim: refactor startup scripts and fix gz sim standalone 
fixes #24598
- refactor px4-rc.simulator into sim specific startup scripts
- fix gz sim launch for PX4_GZ_STANDALONE
- fix gz sim `simulation-gazebo` script by adding the **server.config** file to the models repo
- removed px4-rc.params since it was unused 

Long term we need to have all gz sim stuff in the submodule, with the exception of the gz_bridge
closed 2025-03-25 by dakejahl