PX4/PX4-Autopilot #24608: [Bug] dataman_client unable to write to sd storage

gmesm, 2025-03-25:
### Describe the bug

Starting in PX4 1.15.0, assumed to be with the introduction of https://github.com/PX4/PX4-Autopilot/commit/208552fdab773e19a39f49b8910031f617e11a94, dataman_client is unable to write to SD storage on my tested hardware (EchoPilot AI https://echomav.com/product/echopilot-ai/). Checking dmesg logs and dataman status, datman status is good but dataman_client times out after 5000ms. This issue does not occur on PX4 1.14.4 (before the dataman_client refactor). I have tried restarting dataman after the system has started via a mavlink console but this does not fix the issue. The issue results in unable to write to SD storage errors throughout mission planner and qgroundcontrol. The sd card can be read and written to successfully when tested via mavlink console (eg: `echo thisisatest > /fs/microsd/filename` and `cat /fs/microsd/filename`).

CC: @Igor-Misic

### Vehicle type

None

rosiakpiotr, 2025-06-12:
I am facing similar issue on Pixhawk 6X with PX4 1.16.0 c0. I added custom module implemented as work queue item with `DatamanClient` declared in the module method (the method that is supposed to read all mission items synchronously from the dataman) and it made it impossible reading `mission` with it as it results in `px4_poll` timeouts, seen in `dmesg`. In SITL simulation however the issue dissapears once `DatamanClient` is declared as class field. I suppose the `DatamanClient` constructor has to do with it.

mrpollo, 2025-06-23:
@gmesm @rosiakpiotr, can you please post logs? Make sure you are collecting logs from boot rather than the default mode "starting by arming"; we need to get to the bottom of this and make sure it's not a regression.

@Igor-Misic, are you aware of any issues or have any tips you can share to help us debug?

gmesm, 2025-06-25:
> [@gmesm](https://github.com/gmesm) [@rosiakpiotr](https://github.com/rosiakpiotr), can you please post logs? Make sure you are collecting logs from boot rather than the default mode "starting by arming"; we need to get to the bottom of this and make sure it's not a regression.

Unfortunately I don't have access to the hardware anymore that displayed this issue. Oddly enough I couldn't get the issue to occur on a board from a different batch (no hardware changes) so not really sure what is going on. One of our users has the problem on their board but I don't have a github for them. I'll let them know via another method and maybe they can assist.

ctitus1, 2025-07-03:
Hi everyone, I was the user gmesm was referring to. I'll have to look back to find the logs, but I'm fairly confident I still have them. I should be able to get time in a couple weeks to go digging for this