PX4/PX4-Autopilot #24619: Error in Make File while running px4_sitl gz_x500 command

Tasbiha-mirza, 2025-03-27:
### Describe the bug

I am trying to communicate with Pixhawk using RPi as a companion computer through micro-xrce dds agent and client. I am following this link: [https://docs.px4.io/main/en/ros2/user_guide.html#setup-micro-xrce-dds-agent-client]

When I enter "make px4_sitl gz_x500", I get this error:
ninja: error: unknown target 'gz_x500'
make: *** [Makefile:232: px4_sitl] Error 1

I have tried different solutions, but this error hasn't been solved.
I am using ROS2 Humble, Ubuntu 22.04 and Pixhawk 6C.

### Vehicle type

None

dakejahl, 2025-03-27:
`make px4_sitl gz_x500` is for launching simulation

Officially-Siddhant, 2025-04-05:
@dakejahl is correct. @Tasbiha-mirza given the description provided, you would benefit from using:
`make px4_fmu-v5_default'`
Then run:
`make px4_fmu-v5_default upload`

closed 2025-04-05 by dakejahl