PX4/PX4-Autopilot #24624: [Bug] Error: px4-dev: SHA256 mismatch

beauzo, 2025-03-27:
### Describe the bug

MacOS build workflow is failing due to:
```
Error: px4-dev: SHA256 mismatch
Expected: 6e4a8a6aad592a0c156d16233e16ee276fd6cc63b620be5bd047375b1352e27d
  Actual: 5eda2111dc20c092dc8241768121cf4173edabb593b2b199e6c233b5088c52aa
    File: /Users/runner/Library/Caches/Homebrew/downloads/be4aff9b473b8b0c6bae0ff08a6611885de9d56ee3a477c05453bb6bbcaab246--px4.py
To retry an incomplete download, remove the file above.
```

### Vehicle type

None

onionitsuka, 2025-03-27:
Error: px4-dev: SHA256 mismatch
Expected: 6e4a8a6aad592a0c156d16233e16ee276fd6cc63b620be5bd047375b1352e27d
  Actual: 5eda2111dc20c092dc8241768121cf4173edabb593b2b199e6c233b5088c52aa
  File: /Users/username/Library/Caches/Homebrew/downloads/[...]

Steps I’ve tried so far:
Removing the cached file with:rm /Users/username/Library/Caches/Homebrew/downloads/[...]
brew update
brew upgrade
brew install px4-dev
	Confirming Homebrew is up to date.

Despite these attempts, I’m still getting the same SHA256 mismatch error. Is there a known workaround for this issue, or do we have to wait for a formula update? Any help or suggestions would be greatly appreciated.

Arclunar, 2025-03-28:
same problem

Arclunar, 2025-03-28:
I solved it,  by editing the brew formula manually : 

brew edit px4-dev
in the opened file ,
change the SHA number 6e4a8a6aad592a0c156d16233e16ee276fd6cc63b620be5bd047375b1352e27d  to 5eda2111dc20c092dc8241768121cf4173edabb593b2b199e6c233b5088c52aa

try brew install again . It would work!

beauzo, 2025-03-28:
> I solved it, by editing the brew formula manually :
> 
> brew edit px4-dev in the opened file , change the SHA number 6e4a8a6aad592a0c156d16233e16ee276fd6cc63b620be5bd047375b1352e27d to 5eda2111dc20c092dc8241768121cf4173edabb593b2b199e6c233b5088c52aa
> 
> try brew install again . It would work!

LOL, yeah, that works if you want to take the potentially compromised version.

beauzo, 2025-03-28:
The "real fix" should be updating the `homebrew-px4` repo with the new checksum.

beauzo, 2025-03-28:
Fix: https://github.com/PX4/homebrew-px4/pull/89

beauzo, 2025-03-31:
Issue caused by [this](https://github.com/PX4/PX4-Autopilot/commit/9e99bd3b7a5d4dad3ca68f0137465e18d4834639) commit.

rolandash, 2025-04-04:
This needs to be fixed ASAP.

closed 2025-04-04 by dakejahl