PX4/PX4-Autopilot #24628: [Bug]  Disabling `MPC_ACC_DECOUPLE` can cause vehicle to flip when landing

TedObrien, 2025-03-28:
### Describe the bug

When landing a Helicopter in position mode, it attempted to tip over into the floor upon initial touchdown. Looking at the log you can see a large pitch and roll setpoint is commanded as soon as the ground_contact flag is set to true. The x/y thrust setpoints are also non zero.

[image]

I believe the issue is due to the improper handling of an edge case, introduced in [this commit. ](https://github.com/PX4/PX4-Autopilot/pull/22696) 

When a multicopter is landing, the position controller checks to see if there is ground contact and sets acceleration setpoints to [0,0,100] when detected.

https://github.com/PX4/PX4-Autopilot/blob/19d3e6285be505303e91ba4da4beac6ff291da75/src/modules/mc_pos_control/MulticopterPositionControl.cpp#L509-L513

If `MPC_ACC_DECOUPLE` is disabled the following conditional evaluates to true, and the z- acceleration setpoint is added to `z_specific_force`

https://github.com/PX4/PX4-Autopilot/blob/19d3e6285be505303e91ba4da4beac6ff291da75/src/modules/mc_pos_control/PositionControl/PositionControl.cpp#L209-L212

This is not a problem in normal flight, but when the ground_contact flag is set to 1 by the landing detector, `z_specific_force` becomes massive as maximum negative acceleration is added to it.

```
z_specific_force = -9.8 + 100
```

I think this will massively amplify any non-zero x/y thrust setpoints on the ground (confirmed in SITL). This may be ok for a multicopter without airmode enabled. But I have tested this in sitl with MPC_ACC_DECOUPLE = 0 and MC_AIRMODE = Roll/Pitch (I had to increase the MPC_ACC parameters, but only in SITL) and got the drone to flip when touching down.

**Solution**

I think this could be solved by imposing a limit to how large `z_specific_force` can be, or by checking if the vehicle is landing before adding ` _acc_sp(2)` to `z_specific_force`

### To Reproduce

To get drone to flip over in sitl.
MC_AIRMODE = Roll/Pitch 
MPC_ACC_DECOUPLE = 0
MPC_ACC parameters = 20 m/s^2

Put drone into RTL mode. Drone will landand then flip over https://review.px4.io/plot_app?log=f7b1dd3d-0773-4030-9d63-009e9a26d5ed

https://github.com/user-attachments/assets/0ac60013-774f-441b-bd33-eedfa64b0983

MC_AIRMODE = Roll/Pitch 
MPC_ACC_DECOUPLE = 1
MPC_ACC parameters = 20 m/s^2
Drone will not flip over when landing. https://review.px4.io/plot_app?log=2585fed7-7920-4a1f-9490-1fd7e8721563

https://github.com/user-attachments/assets/40d0e995-07fd-4f5f-a86e-8652671477f8

### Software Version

v1.15 and above

### Vehicle type

Multicopter

### Additional context

Will it be ok to enable MPC_ACC_DECOUPLE for a heli? I had initially disabled it when moving from v1.14 to v1.15 as I wasn't sure how it would affect flight performance,

bresch, 2025-04-04:
Sorry to hear about the crash. I didn't realize the vertical acceleration is set to 100 when landed...

TedObrien, 2025-04-07:
Hi @bresch, the pilot managed to go into stabilized mode and save it, so no crash. It was very exciting though!

I believe changing line https://github.com/PX4/PX4-Autopilot/blob/19d3e6285be505303e91ba4da4beac6ff291da75/src/modules/mc_pos_control/PositionControl/PositionControl.cpp#L209

to:
```	
if (!_decouple_horizontal_and_vertical_acceleration && !flying_but_ground_contact ) {
```

Will solve this issue. I can make a pr if its helpful?

closed 2026-08-28 by github-actions[bot]