PX4/PX4-Autopilot #24631: [Bug] SITL configuration fails on gazebo plugins even if I'm not using gazebo at all

MaEtUgR, 2025-03-28:
### Describe the bug

When I run `make px4_sitl` on main I get the following configuration error:
```
CMake Error at src/modules/simulation/gz_plugins/optical_flow/CMakeLists.txt:42 (add_library):
  Target "OpticalFlowSystem" links to target "gz-sensors::gz-sensors" but the
  target was not found.  Perhaps a find_package() call is missing for an
  IMPORTED target, or an ALIAS target is missing?
```

Simulating with `make px4_sitl gz_x500` used to work fine until recently, now it doesn't work anymore. But I didn't try to use gazebo just build SITL and run it with SIH and jmavsim which still works if I disable GZ plugins here:
https://github.com/PX4/PX4-Autopilot/blob/72454c4fd24c7047f56454d25218d0ba1c36068b/boards/px4/sitl/default.px4board#L56

### Expected behavior

Builds that don't use Gazebo should not fail because of Gazebo.

Why Gazebo suddenly doesn't work anymore, possibly version not supported anymore can be sorted separately.

### Screenshot / Media

[image]

### Software Version

I have Ubuntu 22.04 and
```
$ gz sim --versions
7.6.0
```

### Flight controller

SITL

### Vehicle type

None

MaEtUgR, 2025-03-28:
I made an automatic bisection search and d2c049ec93f8f30901cfaecd9fbef2c77a7b9199 broke the SITL build without gazebo.
@dakejahl sorry to bother you but do you have an idea? 😇

MaEtUgR, 2025-03-28:
It's the `gz_plugins/CMakeLists.txt` and I'm in the upper version case:
https://github.com/PX4/PX4-Autopilot/pull/24518/files#diff-831e237e31eac3333af54b6c8b5b8944c54ffa13ae8a48830468719a283518feR37-R40

MaEtUgR, 2025-03-28:
I can't point my finger at it but calling `add_subdirectory(optical_flow)` without me having the right gezbo version seems to crash it. I guess gz transport is found but the version doesn't matter or something like that 🤔

MaEtUgR, 2025-03-28:
If I remove `gz-transport12` from the `find_package()` list it also builds. Are we not compatible with that version? 🤔 Or am I digging in the wrong corner?

MaEtUgR, 2025-03-28:
I might be up to something. At least before that change gz-transport12 was not a possible option CMake was looking for:
[image]
So either we can't support gz-transport12 anymore or something broke before that which had still supported it.

dakejahl, 2025-03-28:
Harmonic is the minimum supported version, which is transport13
https://github.com/gazebo-tooling/gazebodistro/blob/master/collection-harmonic.yaml

I think I should just revert some parts of this commit so bring back the better error handling
https://github.com/PX4/PX4-Autopilot/commit/fced29da598e496a6ab3f19b6529ae97c2081b8a

MaEtUgR, 2025-03-28:
Aha, only the plugins look for transport 12 https://github.com/PX4/PX4-Autopilot/commit/fced29da598e496a6ab3f19b6529ae97c2081b8a#diff-831e237e31eac3333af54b6c8b5b8944c54ffa13ae8a48830468719a283518feR33-R37
, gz_bridge doesn't
https://github.com/PX4/PX4-Autopilot/commit/fced29da598e496a6ab3f19b6529ae97c2081b8a#diff-41d28c50b16305ce484c347ce129510d199bcf103616869a0dccd93310a53656R35-R36

I'm quickly checking when my gazebo stopped working 👀

MaEtUgR, 2025-03-28:
I think I figured it out @perrrre (former @Perrrewi) added tailsitter support in https://github.com/PX4/PX4-Autopilot/pull/24028 which breaks the Gazebo garden (which I apparently still have) build because of some airspeed dependency and then somewhat shortly after removed Gazebo garden from being allowed in https://github.com/PX4/PX4-Autopilot/pull/24089 (which I have even merged myself 😇):
[image]

But the bottom line is transport 12 is not supported anymore like you wrote so I'll suggest that.

closed 2025-03-28 by dakejahl

dakejahl, 2025-03-28:
No problem! Do we still need to add some better error messages for the next guy who hits this problem?