PX4/PX4-Autopilot #24639: [Bug] Pixhawk 6X&6X Pro ESC Calibration Error

HB-SP-P, 2025-03-31:
### Describe the bug

Unable to calibrate ESC with ESC Calibration failed popup warning user to unplug battery first even if the flight controller is powered by USB.

### To Reproduce

1. Connect FC to PC via USB
2. Open QGC
3. In the Power tab under ESC PWM calibration click Calibrate
4. ESC Calibration failed popup

### Expected behavior

After clicking the Calibrate, there should be a warning about removing the propellers and waiting for battery power.
The calibration process was normal using 1.13.3 firmware.

### Screenshot / Media

[image]

[image]

### Software Version

PX4 version 1.15.4 & 1.15.0

### Flight controller

Pixhawk 6X&6X Pro

### Vehicle type

Multicopter

### How are the different components wired up (including port information)

Only the USB was connected

### Additional context

Have a customer report changing flight mode to manual will fix this issue, but was not able to reproduce the fix.

amovlgf, 2025-03-31:
The battery voltage is not calibrated

HB-SP-P, 2025-04-01:
Correct screenshot:
[image]

YHuniv, 2025-04-01:
try the solution in this link : 
[ESC calibration Manual mode](https://github.com/PX4/PX4-Autopilot/issues/24401)

PR #24800 (2025-06-04): Fix reporting of connected battery
Fixes #24401
Fixes #24645
Fixes #24639

I haven't been able to verify this since I can't seem to build a firmware on Mac arm. But I believe this fixes the problem. Hopefully someone else can verify. And then this needs to get into v1.16 as well however that happens.