PX4/PX4-Autopilot #24645: [Bug] Possible ESC calibration issue

afwilkin, 2025-03-31:
### Describe the bug

Discovered an ESC cal issue with a branch of 1.16. 

This ticket here is just to track it. Will flesh out more in a bit

### Vehicle type

None

DroneWuKong, 2025-04-02:
Just ran this and i'm not seeing the issue via mainline, using QGC Daily and Main.

Log for Main: https://review.px4.io/plot_app?log=a2098618-b574-4ae2-b4a0-e9e7ba45b33d

Log for Daily: https://review.px4.io/plot_app?log=77ec7482-0c7e-42cb-8298-6c1a4ad729c0

afwilkin, 2025-04-08:
I am seeing this when I am trying to calibrate ESCs. I do not power on the drone at all, and try to calibrate out of the box

[image]

I am seeing an issue when trying to calibrate 6c and NXP flight controllers. 

Not the EXACT same issue, but an issue

[NXP](https://www.loom.com/share/882d61a714554ba5bc06d10cdeb6e422)
[Holybro 6C](https://www.loom.com/share/13b52fdf5f4543eeb110b97a64039af4?sid=b815a2c3-d50a-49ad-bb77-81039e1dccb1)

afwilkin, 2025-04-08:
ok, that is what is wrong. The esc calibration appears to fail in esc_calibration.cpp somewhere in "check_battery_disconnected" function. 

it works when you

1. connect drone over USB
2. power on drone
3. power off drone (without disconnecting usb)
4. start esc calibration as normal. This should now work

afwilkin, 2025-04-08:
https://github.com/PX4/PX4-Autopilot/blob/e228456a67bfeef247ea451421fdf6dbee9eee3a/src/modules/commander/esc_calibration.cpp#L64

afwilkin, 2025-04-08:
digging into the specific code now

afwilkin, 2025-04-08:
Yeah this is conditional that is failing: 

https://github.com/PX4/PX4-Autopilot/blob/e228456a67bfeef247ea451421fdf6dbee9eee3a/src/modules/commander/esc_calibration.cpp#L66

afwilkin, 2025-04-08:
I just flashed 1.15.4 and this appears to ALSO be an issue on 1.15.4. 

Going to try on holybro 6c

afwilkin, 2025-04-08:
Pixhawk 6c power module: PM02 V3.2
NXP power module: PM02D-v1.4

So, I THINK there are 2 issues

1. the driver for these 2 power modules is different. the PM02 V3.2 seems to publish a battery message when no battery is plugged in, while the driver for PM02D-v1.4 does NOT
2. When PM02 V3.2 is plugged in, there is not an added check to make sure that a battery is plugged in before continuing the ESC calibration

PetervdPerk-NXP, 2025-04-09:
> the driver for these 2 power modules is different. the PM02 V3.2 seems to publish a battery message when no battery is plugged in, while the driver for PM02D-v1.4 does NOT

I think this is correct a 6C get battery status from ADC on the 6C itself thus is always active.
An 6X/6X-RT receives it through i2c using a ina2xx module, if there's no digital power module connected then the ina2xx will not publish a battery message.

dakejahl, 2025-05-04:
What's the status here? I'm looking at our release blockers/issues
https://github.com/orgs/PX4/projects/45/views/11

dakejahl, 2025-05-04:
https://github.com/PX4/PX4-Autopilot/pull/24800

PR #24800 (2025-06-04): Fix reporting of connected battery
Fixes #24401
Fixes #24645
Fixes #24639

I haven't been able to verify this since I can't seem to build a firmware on Mac arm. But I believe this fixes the problem. Hopefully someone else can verify. And then this needs to get into v1.16 as well however that happens.