PX4/PX4-Autopilot #24651: [Bug] MAVSDK tests need to be updated to work with Gazebo

hamishwillee, 2025-04-01:
### Describe the bug

The MAVSDK tests are run in CI using gazebo classic, and the docs also require gazebo classic: https://docs.px4.io/main/en/test_and_ci/integration_testing_mavsdk.html

However as we're moving to Ubuntu versions that don't support Gazebo Classic, if we want these tests we'll need them to run with Gazebo.

Note, currently we get this error on Ubuntu 22.04:

```
ubuntu@DESKTOP-QN38EN3:~/github/PX4-Autopilot$ make px4_sitl gz_x500 mavsdk_tests
ninja: error: unknown target 'mavsdk_tests'
make: *** [Makefile:232: px4_sitl] Error 1
````

### Vehicle type

None

hamishwillee, 2025-04-01:
FYI, added @julianoes and @mrpollo to this as an FYI, and because you're the ones working on MAVSDK, updating the toolchain.

mrpollo, 2025-04-18:
@julianoes if you have the time ping me, I can spare funds for this effort.

mrpollo, 2025-04-29:
Thanks @julianoes, when you have the time ping me, and we can schedule a call.

dakejahl, 2025-07-10:
@julianoes @mrpollo where did we end up with this?

julianoes, 2025-07-11:
So we have updated mavsdk itself, but the tests are still with old Gazebo. So the next step would be to switch the tests over to new Gazebo...