PX4/PX4-Autopilot #24654: [Bug] Hover thrust estimator bug

dakejahl, 2025-04-02:
### Describe the bug

When flying in Manual mode, shortly after takeoff the drone rapidly loses altitude. This coincides with `hover_thurst_estimate.valid`. I set MPC_USE_HTE to 0 and this continues to occur. It seems like the throttle setpoint changes for some reason.

https://review.px4.io/plot_app?log=2d4d8f06-edca-4e79-a5b9-5c729d971e15

[image]

### Vehicle type

None

MaEtUgR, 2025-04-09:
Related? Hover thrust is now considered for the stick scaling in Stabilized/Manual mode: https://github.com/PX4/PX4-Autopilot/pull/19633

bresch, 2025-05-02:
fixed by https://github.com/PX4/PX4-Autopilot/pull/24710

closed 2025-05-02 by bresch