PX4/PX4-Autopilot #24659: [Bug] Unable to setup Windows Development Environment to launch simulation

skiff2011, 2025-04-02:
### Describe the bug

Following [documentation](https://docs.px4.io/main/en/dev_setup/dev_env_windows_wsl.html) to setup PX4 development environment on Windows  11 running on Ubuntu 22.04 Linux within WSL2. 
`make px4_sitl`
completes with error

user@HomeDesktop:~/PX4-Autopilot$ make px4_sitl
-- PX4_GIT_TAG: v1.16.0-beta1-56-gdb97dd471d
-- Found PythonInterp: /usr/bin/python3 (found suitable version "3.10.12", minimum required is "3")
-- PX4 config file: /home/user/PX4-Autopilot/boards/px4/sitl/default.px4board
-- PLATFORM posix
-- ROMFSROOT px4fmu_common
-- TESTING y
-- ETHERNET y
-- ROOT_PATH .
-- PARAM_FILE /fs/mtd_params
-- PX4 config: px4_sitl_default
-- PX4 platform: posix
-- PX4 lockstep: enabled
-- The CXX compiler identification is GNU 11.4.0
-- The C compiler identification is GNU 11.4.0
-- The ASM compiler identification is GNU
-- Found assembler: /usr/bin/cc
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Check for working CXX compiler: /usr/bin/c++ - skipped
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working C compiler: /usr/bin/cc - skipped
-- Detecting C compile features
-- Detecting C compile features - done
-- cmake build type: RelWithDebInfo
-- ccache enabled (export CCACHE_DISABLE=1 to disable)
-- Looking for gz-transport13 -- found version 13.4.1
-- Searching for dependencies of gz-transport13
-- Found ZLIB: /usr/lib/x86_64-linux-gnu/libz.so (found version "1.2.11")
-- Looking for pthread.h
-- Looking for pthread.h - found
-- Performing Test CMAKE_HAVE_LIBC_PTHREAD
-- Performing Test CMAKE_HAVE_LIBC_PTHREAD - Success
-- Found Threads: TRUE
-- Checking for module 'uuid'
--   Found uuid, version 2.37.2
-- Found UUID: TRUE
-- Looking for gz-utils2 -- found version 2.2.1
-- Searching for dependencies of gz-utils2
-- Searching for <gz-utils2> component [cli]
-- Looking for gz-utils2-cli -- found version 2.2.1
-- Searching for dependencies of gz-utils2-cli
-- Looking for gz-msgs10 -- found version 10.3.2
-- Searching for dependencies of gz-msgs10
-- Looking for gz-utils2 -- found version 2.2.1
-- Looking for gz-math7 -- found version 7.5.1
-- Searching for dependencies of gz-math7
-- Looking for gz-utils2 -- found version 2.2.1
-- Checking for module 'tinyxml2'
--   Found tinyxml2, version 9.0.0
-- Found Python3: /usr/bin/python3.10 (found version "3.10.12") found components: Interpreter
-- Found Protobuf: /usr/lib/x86_64-linux-gnu/libprotobuf.so (found version "3.12.4")
-- Looking for gz-transport13 -- found version 13.4.1
-- Searching for dependencies of gz-transport13
-- Checking for module 'uuid'
--   Found uuid, version 2.37.2
-- Looking for gz-utils2 -- found version 2.2.1
-- Searching for dependencies of gz-utils2
-- Searching for <gz-utils2> component [cli]
-- Looking for gz-utils2-cli -- found version 2.2.1
-- Searching for dependencies of gz-utils2-cli
[... 279 lines cut ...]
compilation terminated due to -Wfatal-errors.
[29/1082] Generating uORB topic ucdr headers
ninja: build stopped: subcommand failed.
make: *** [Makefile:232: px4_sitl] Error 1

### To Reproduce

performed
`git clone https://github.com/PX4/PX4-Autopilot.git --recursive`
`bash ./PX4-Autopilot/Tools/setup/ubuntu.sh`
then restarted WSL
`make px4_sitl`
ends with error

### Expected behavior

`make px4_sitl` finishes without errors

### Software Version

Windows 11 WSL2 Ubuntu 22.04

### Vehicle type

None

dakejahl, 2025-04-02:
which version of protoc is installed?
```
protoc --version
```

skiff2011, 2025-04-02:
after WSL installation, there is no protoc. After launching 
`bash ./PX4-Autopilot/Tools/setup/ubuntu.sh`
it shows 3.12.4

skiff2011, 2025-04-02:
also tried installing 3.6.1 prior px4 setup - didnt help

dakejahl, 2025-04-02:
3.12.4 is the same version I have so it should be fine

What about gz version?
```
gz sim --versions
```

skiff2011, 2025-04-02:
```
gz sim --versions
8.9.0
```

hamishwillee, 2025-04-03:
I just tried this with a fresh Ubuntu 22.04.5 WSL2 installation from the store on Windows 11. Everything installed cleanly. I typed `exit` at the end to shut down the powershell, then restarted a shell with WSL2 and ran `make px4_sitl` and `make px4_sitl gz_x500`. Both worked fine.

For me this is "cannot reproduce".

skiff2011, 2025-04-03:
tried once more - still no luck. Maybe I'll try on fresh windows install

dakejahl, 2025-04-03:
I think @AlexKlimaj uses WSL without any issues?

AlexKlimaj, 2025-04-03:
I don't do SITL in WSL. Only building firmware.

skiff2011, 2025-04-03:
After windows reinstall everything works. Anyway thanks for help

closed 2025-04-03 by skiff2011