PX4/PX4-Autopilot #24665: Airspeed Sensor Failure on PX4 with Cube Orange During Flight

themboutidem, 2025-04-03:
### Describe the bug

On multiple short flights using a Sensirion SDP3x Airspeed Sensor with the Cube Orange Flight Controller running a version of PX4 Autopilot v1.15.1, the airspeed sensor fails a few seconds into the flight. 

Error Messages in Logs:
CRITICAL: Airspeed sensor failure detected. Return to launch (RTL) is advised
WARNING: [health_and_arming_checks] Preflight Fail: Airspeed invalid.

### To Reproduce

- Power on the Drone
- Connect the SDP3x sensor to I2C 2 port.
- Upload a mission using QGroundControl.
- Send Takeoff command from QGroundControl
- Monitor QGC for airspeed sensor failure messages

### Expected behavior

- The airspeed sensor should provide continuous readings throughout the flight. 

### Flight Log

- Log 1: https://review.px4.io/plot_app?log=2d6237c7-3eb9-4a1a-b3e5-101f365aa7de
- Log 2: https://review.px4.io/plot_app?log=4e08169b-9eac-472c-8cd9-30d376a130d4
- Log 3: https://review.px4.io/plot_app?log=e226e3e1-892d-428f-b6b4-ff06aec3bfd2

### Software Version

PX4 v1.15.1 (with minor modifications to include a custom driver for a new sensor)

### Flight controller

Cube Orange+ 

### Vehicle type

Fixed Wing

TruAero, 2025-05-02:
I am having a similar issue with the MS4525DO Digital Air Speed Sensor on a Pixhawk 6X.

ryanjAA, 2025-05-05:
on both above, it is more likely you have settings incorrect (or a clogged pitot).

On SDP units you need to enable tube pressure drop and also input the tubing length and diameter, not forgetting the extra length in the pitot itself.

in theMS4525 you would get a fix by swapping the autopilot, it just reads data from the sensor. 

@themboutidem more info is needed.

ryanjAA, 2025-05-06:
That by definition is emi. I2c is super susceptible to it and even if using a can converter, it’s still i2c in. If you changed the 4425 board itself I wouldn’t be skeptic but unless you have a broken i2c port, the autopilot wasn’t the root cause (barring you hadn’t crashed it or there was a fundamental hardware issue with it somewhere). Otherwise, it was either a firmware (are you running anything custom?), lurking setting, or physical change. Nothing else makes deterministic sense, meaning changing the autopilot and have that be the fix isn’t a fix (if it were an onboard sensor then yes) - something else was occurring.

Upload a before and after log.

TruAero, 2025-05-06:
I can’t share the logs unfortunately but yes I did change the 4425 board. I agree replacing the flight controller was probably not the direct fix to the issue. I’m not using any custom firmware.

TruAero, 2025-05-08:
> That by definition is emi. I2c is super susceptible to it and even if using a can converter, it’s still i2c in. If you changed the 4425 board itself I wouldn’t be skeptic but unless you have a [...]

We saw this issue again in flight. After twisting the MS4425 I2C cable and slightly moving some power cables, the issue did not repeat in subsequent flights. Potentially confirming your EMI suspicions.

ryanjAA, 2025-05-08:
> We saw this issue again in flight. After twisting the MS4425 I2C cable and slightly moving some power cables, the issue did not repeat in subsequent flights. Potentially confirming your EMI [...]

Open a separate issue. Completely different sensor.