PX4/PX4-Autopilot #24741: [Bug] Preflight check: FAILED without further information

christianrauch, 2025-04-21:
### Describe the bug

`commander check` returns `Preflight check: FAILED` without further information why the check fails.

QGroundControl reports an "X/Y position control Error" and all other sensors as "Normal".

### To Reproduce

`commander check`

### Expected behavior

No issues or information about the reason for the failure.

### Software Version

```
HW arch: PX4_RASPBERRYPI
PX4 git-hash: 1ea672e87d1d6c5d03ecee6ff2fbe616e6059288
PX4 version: 1.16.0 80 (17825920)
PX4 git-branch: main
OS: Linux
OS version: Release 6.8.0 (101187839)
Build datetime: Apr 21 2025 12:50:09
Build uri: localhost
Build variant: default
Toolchain: GNU GCC, 13.3.0
PX4GUID: 100449495052303030443030303030303030
UNKNOWN MCU
```

### Flight controller

Raspberry Pi 5

### Vehicle type

Multicopter

### How are the different components wired up (including port information)

pca9685, mpu9250, ak8963 via I2C

sfuhrer, 2025-04-30:
Could you share a log file by chance? You can create one without flying through the MAVLink shell:
```
logger on 
logger off
```
then download the log and share it here.

christianrauch, 2025-05-03:
I am unable to log to file:
```
ERROR [logger] failed creating log root dir: /fs/microsd/log (2)
```
and it does not seem possible to change the file path.

With
```
logger start -m mavlink
```
I don't see log messages in the MAVLink Inspector in QGroundControl.

christianrauch, 2025-05-04:
I could only find the `X/Y position control` string in QGroundControl:
https://github.com/mavlink/qgroundcontrol/blob/ace8f62c730d9c1553b8b4e2d33cd24982a7eed9/src/MAVLink/QGCMAVLink.cc#L285

The `MAV_SYS_STATUS_SENSOR_XY_POSITION_CONTROL` relates to `health_component_t::local_position_estimate` in PX4 and I could finally track this down to:
https://github.com/PX4/PX4-Autopilot/blob/38d67f5a937f380e36cdeb8f90884e20ab1b683f/src/modules/commander/HealthAndArmingChecks/checks/modeCheck.cpp#L83-L85

It is suboptimal that you have to crawl through two code bases to get a hint on what the FC is doing. Still, there is no information presented that explains where this issue originates from.