PX4/PX4-Autopilot #24758: [Bug] The body angular velocity (body rate) values published on the ROS 2 topic /fmu/out/vehicle_angular_velocity differ significantly from the values shown in QGroundControl (QGC) via MAVLink messages.

tuananh0606-tech, 2025-04-24:
### Describe the bug

The body angular velocity (body rate) values published on the ROS 2 topic /fmu/out/vehicle_angular_velocity differ significantly from the values shown in QGroundControl (QGC) via MAVLink messages. This discrepancy causes confusion during controller debugging and system analysis.

### To Reproduce

1. Run ```make px4_sitl gz_x500``` in terminal PX4-Autopilot.
2. Run ```MicroXRCEAgent udp4 -p 8888``` in other terminal.
3. Run ```ros2 topic echo /fmu/out/vehicle_angular_velocity``` in a terminal within the ros_ws that contains px4_msgs.
4. At the same time, monitor body angular velocity (e.g., gyro rates) in QGroundControl through the MAVLink Inspector.
5. Compare values between the ROS 2 topic and QGC – they show noticeable differences.

### Expected behavior

The angular velocity (body rate) values from /fmu/out/vehicle_angular_velocity should closely match those shown in QGroundControl, as both should represent the same onboard state estimation.

### Screenshot / Media

[image]

### Flight Log

https://docs.px4.io/main/en/msg_docs/VehicleAngularVelocity.html

### Software Version

PX4 version: v1.15.2

ROS 2: Humble

PX4-ROS2 Bridge: uXRCE v2.4.2

Simulation: Gazebo-harmonic

### Vehicle type

None

### Additional context

I am currently developing a custom controller that relies on accurate body rate feedback via ROS 2. The mismatch with QGC values raises concerns about which source is accurate, or whether a transformation (e.g., NED to FLU or timestamp alignment) is needed.

beniaminopozzan, 2025-04-27:
Hi @tuananh0606-tech , The `ATTITUDE` MAVLink message uses exactly the `vehicle_angular_velocity` `xyz` filed to get `rollspeed`, `pitchspeed` and `yawspeed`.
The data does not match because you are seing messages captured at different timestamps.

Try matching the timestamps (you can turn off the time synchronized of uxrcedds so that the ROS 2 messages will keep the PX4 timestamp and make the matching agints the MAVLink timestamp easier), you will see the value to to match too (well, only if you get an **exact** time match).

tuananh0606-tech, 2025-04-28:
@beniaminopozzan but i still get correct information about position, velocity and angle, only angular velocity is not correct

beniaminopozzan, 2025-04-29:
@tuananh0606-tech _correct_ as the data matches **perfectly**, not a single digit is off?
I might expect that from pos and vel as the publication rate is low enough that it is likely that you see the same messge but I won't expect this from attitude. Maybe it looks correct just because the data is close enough.
I would argue that even the angular velocity is correct as the difference between ROS2 and MAVLink is in the order of 1e-3 rad/s

tuananh0606-tech, 2025-04-30:
@beniaminopozzan Do I need to convert to another coordinate frame?

beniaminopozzan, 2025-05-17:
no conversion is needed if you compare raw MAVLink and urceDDS

Conversion is needed if you use MAVROS instead as it perform NED->ENU and FRD->FLU