PX4/PX4-Autopilot #24770: [Bug] [uxrce_dds_client] disconnected

crow-owl, 2025-04-28:
### Describe the bug

I am currently setting up communication between Pixhawk and ROS 2 via uXRCE-DDS.
Despite reviewing numerous official documents and forum discussions, the issue remains unresolved in my case.
I consistently **fail to connect** the uxrce_dds_client. Below is the result:

```
nsh> uxrce_dds_client start -t udp -b 921600 -p 8888
INFO  [uxrce_dds_client] init UDP agent IP:192.168.50.112, port:8888
nsh> uxrce_dds_client status
INFO  [uxrce_dds_client] Running, disconnected
INFO  [uxrce_dds_client] Using transport:     udp
INFO  [uxrce_dds_client] Agent IP:            192.168.50.112
INFO  [uxrce_dds_client] Agent port:          8888
INFO  [uxrce_dds_client] Custom participant:  no
INFO  [uxrce_dds_client] Localhost only:      no
INFO  [uxrce_dds_client] timesync converged: false
uxrce_dds_client: cycle: 0 events, 0us elapsed, 0.00us avg, min 0us max 0us 0.000us rms
uxrce_dds_client: cycle interval: 0 events, 0.00us avg, min 0us max 0us 0.000us rms
```

### To Reproduce

Before the description, I will provide my hardware setup.

# Hardware
|Name|SPEC|
|---|---|
|Drone|Holybro S500 V2|
|Companion Computer|Intel NUC|
|OS|Ubuntu 22.04.5 LTS|
|Flight Controller|Pixhawk 6X|
|Firmware|	PX4 v1.15.4|
|Connection|Ethernet|

# Board Connection

[image]

# Parameter

[image]

[image]

[image]

[image]

# Parameter Intend

I want to connect the Pixhawk and the NUC via Ethernet using the uXRCE-DDS protocol.
Therefore, I connected the Pixhawk and the NUC through the Ethernet port.
Additionally, I connected the Pixhawk and the NUC via USB-C to supply power and to connect with QGroundControl (QGC).

As a result, I configured the parameters as follows:

- MAV_0_CONFIG = TELEM1

- MAV_1_CONFIG = Disabled

- MAV_2_CONFIG = Disabled

Here, MAV_0_CONFIG is used for the QGC connection.
UXRCE_DDS_CFG is set to Ethernet.
UXRCE_DDS_AG_IP is set to the Ethernet IP address (192.168.50.112).
UXRCE_DDS_DOM_ID is set to 30, which is the common ROS Domain ID used in our laboratory.

Note that SER_TEL1_BAUD is set to 57600 8N1, but the parameters SER_TEL0_BAUD and SER_TEL2_BAUD do not appear.

[... 222 lines cut ...]
nsh> ver all
HW arch: PX4_FMU_V6X
HW type: V6X
HW FMUM ID: 0x008
HW BASE ID: 0x000
PX4 git-hash: 99c40407ffd7ac184e2d7b4b293f36f10fe561ef
PX4 version: Release 1.15.4 (17761535)
OS: NuttX
OS version: Release 11.0.0 (184549631)
OS git-hash: 5d74bc138955e6f010a38e0f87f34e9a9019aecc
Build datetime: Mar 20 2025 23:04:15
Build uri: localhost
Build variant: default
Toolchain: GNU GCC, 9.3.1 20200408 (release)
PX4GUID: 000600000000383832353233510c003c0045
MCU: STM32H7[4|5]xxx, rev. V
```

### Flight controller

Pixhawk 6X

### Vehicle type

Multicopter

mrpollo, 2025-04-28:
Sounds like you have networking problems, try doing a ping to the NUC from something else in the network (not Pixhawk) like your laptop. If this doesn't work then you have to fix whatever is wrong in your NUC networking configuration, most likely a firewall issue.

crazy-robotics, 2025-04-29:
Check network routing on `NUC`

crow-owl, 2025-04-29:
@mrpollo 
Thanks for the hint about basic networking.

I verified ICMP both ways:

**NUC ➜ laptop** 
```
$ ping -c 4 192.168.50.103  
→ 0 % packet loss
```

**laptop ➜ NUC**  
```
$ ping -c 4 192.168.50.92  
→ 0 % packet loss
```

Aslo ufw is inactive in NUC
```
cowboy@GCSL-NUC1:~$ sudo ufw status
Status: inactive
```

crow-owl, 2025-04-30:
[image]

# Linakge Problem

Recently, I found that the communication between the NUC and PX4 is unstable.
The Ethernet port LED is blinking, but enp2s0 shows **"no link."**

Any ideas on how to resolve this Ethernet link issue?

YHuniv, 2025-04-30:
**-b 921600** sould be used for serial Com not udp. 
for udp : try **uxrce_dds_client start -t udp -p 8888**

crow-owl, 2025-04-30:
> **-b 921600** sould be used for serial Com not udp. for udp : try **uxrce_dds_client start -t udp -p 8888**
> 
> If you start MicroXRCEAgent udp4 -p 8888 in your compagnon computer tehre is no need to do it in FC too.
> 
> iIn MAVLink Console (Pixhawk Side) : Be sur to use IP adresse from your network , in your case Agent IP: 192.168.50.112 so do : **ifconfig eth0 192.168.50.2**
> 
> Be sur taht **eht0** is thhe good name, by :
> 
> ```
> ifconfig
> 
> eth0    Link encap:Ethernet HWaddr xx:xx:xx:xx:xx:xx at UP
>         inet addr:169.254.21.184 DRaddr:169.254.21.1 Mask:255.255.255.0
> ```

**Thanks for your comment.**

After some additional checks this is what I see on each side:

**Pixhawk (NSH)**  
```text
nsh> ifconfig
eth0    Link encap:Ethernet  HWaddr 66:55:11:d7:99:ce  at DOWN
        inet addr:192.168.50.2  DRaddr:192.168.50.92  Mask:255.255.255.0
```

**NUC**  
```text
2: enp86s0: <NO-CARRIER,BROADCAST,MULTICAST,UP> mtu 1500 qdisc mq state DOWN group default qlen 1000
    link/ether 48:21:0b:60:55:95 brd ff:ff:ff:ff:ff:ff
```

The cable is plugged in, but the link is still down. This seems very close to the situation described in [[PX4/PX4-Autopilot #22558](https://github.com/PX4/PX4-Autopilot/issues/22558)].

I’m therefore debating two options:

* **Try a new, non-PoE Ethernet cable**  
* **Switch to a Pixhawk 6C and use pure serial instead**

If you have experience with a robust *wired* setup for ROS 2 topic transfer between a companion computer and Pixhawk, I’d really appreciate your advice. My intended layout is:

| Link | Purpose |
|------|---------|
| **Pixhawk → NUC: FTDI-UART cable** | uXRCE-DDS data |
| **Pixhawk → NUC: USB-C ↔ USB-A** | Power + QGroundControl |

Does this arrangement make sense, or is there a better practice?

Thanks again for your help!

crow-owl, 2025-04-30:
> Check network routing on `NUC`

@crazy-robotics Thanks for your comment!

**Routing check on the NUC**

```
cowboy@GCSL-NUC1:~$ ip -4 route
default via 192.168.50.1 dev wlo1 proto dhcp metric 600
169.254.0.0/16 dev wlo1 scope link metric 1000
192.168.50.0/24 dev wlo1 proto kernel scope link src 192.168.50.92 metric 600
```

As you can see, all traffic for 192.168.50.0/24 is still going through the Wi-Fi interface wlo1, not the Ethernet interface enp86s0, so it does look like a routing problem on the NUC.

I followed every step in the PX4 Ethernet-setup guide:
https://docs.px4.io/main/en/advanced_config/ethernet_setup.html

But the issue remains.

At the moment I’m trying the workaround described in this issue: 
https://github.com/PX4/PX4-Autopilot/issues/22558

If anyone has additional ideas, I’d really appreciate the help.
Thanks in advance!

mrpollo, 2025-04-30:
I did a double check of the networking details and they look sane. The only thing i can think of is maybe try `netman update`. if you changed the networking details via mavlink console, you should always run `netman update` afterwards, unfurtunately `netman save` is poorly named, it doesn't actually publish the information to the system, instead it saves them to a file in the filesystem for you to edit later.

crow-owl, 2025-05-06:
@mrpollo 
Thanks a lot for checking and suggesting `netman update`. I gave it a try, and while it seems to work properly, I believe the root cause of the issue might be something else. I'm still seeing the same symptoms after applying the update. Any further suggestions would be really appreciated!

YHuniv, 2025-05-06:
Pixhawk (NSH)

nsh> ifconfig
```
eth0    Link encap:Ethernet  HWaddr 66:55:11:d7:99:ce  at **DOWN**
        inet addr:192.168.50.2  DRaddr:192.168.50.92  Mask:255.255.255.0
```
NUC

2```
: enp86s0: <NO-CARRIER,BROADCAST,MULTICAST,UP> mtu 1500 qdisc mq state **DOWN** group default qlen 1000
    link/ether 48:21:0b:60:55:95 brd ff:ff:ff:ff:ff:ff
```

I  saw that for your **eth0** **and enp86s0** bought are down. 
you can try : 

`sudo ip link set dev eth0` up or` ifup eth0` do same for `enp86s0`

closed 2026-01-14 by github-actions[bot]