PX4/PX4-Autopilot #24772: [Bug] TrajMath.h: Increasing acceleation increases the braking distance

Luka-Filipovic, 2025-04-28:
### Describe the bug

The function that returns the maximum speed from the braking distance and jerk/accel is an approximation used to easily determine the braking speed. The problem with it is that increasing acceleration slows down the speed.
```
inline float computeMaxSpeedFromDistance(const float jerk, const float accel, const float braking_distance,
		const float final_speed)
{
	auto sqr = [](float f) {return f * f;};
	float b =  4.0f * sqr(accel) / jerk;
	float c = - 2.0f * accel * braking_distance - sqr(final_speed);
	float max_speed = 0.5f * (-b + sqrtf(sqr(b) - 4.0f * c));

	// don't slow down more than the end speed, even if the conservative accel ramp time requests it
	return fmaxf(max_speed, final_speed);
}
```

A plot that shows the function output, speed in function of distance, for different accelerations.
[image]

A plot that shows the speed profile when braking 10 meters for different accelerations.
[image]

### Vehicle type

None

dakejahl, 2025-05-04:
I think this is the issue
> We assume a constant acceleration profile with a delay of 2*accel/jerk

This assumes our initial acceleration is the max accel value, so as we increase the max accel value we're also increasing the distance derived in the jerk delay term. Instead of assuming initial acceleration, should we instead pass in the current acceleration? @bresch can you review?

jkflying, 2025-05-08:
Correct, this makes a conservative worst-case assumption that you are accelerating at your max acceleration towards the target point. It can be considerably more aggressive if you start from a zero acceleration assumption, or best of both would be if you can indeed pass in the current acceleration.

Been a while 😄

Saibernard, 2026-10-05:
@dakejahl are you working on this one? If not, I can take it if you want.

I worked out the exact stop for a jerk limited profile that starts at the current acceleration, ramps to the deceleration limit and back to zero, and checked it against a simulation. From zero acceleration the distance is v²/2a + va/2j once the limit is reached, and v·sqrt(v/j) below that. The code solves v²/2a + 2av/j, so at the MC auto defaults a stop from 5 m/s gets 11.7 m where 6.0 m is enough, and for a 10 m stop the allowed speed drops from 4.6 to 2.4 m/s as the limit goes from 2 to 8 m/s².

What I had in mind is passing the current acceleration as a fifth argument, like you mentioned, and using it from the generators that already have it (position smoothing, TECS, follow target, land nudging). Collision prevention and the rovers don't have it, so I'd keep collision prevention on a worst case and assume zero for the rovers, unless you'd rather do those differently.

Saibernard, 2026-10-05:
@dakejahl a follow up after flying it. I tried the exact version in SIH, quadx flying out and back with a stop at the end, compared with current main. This is how far past the final stop it ends up, negative means it stops short, and the vehicle numbers held within a few cm over three runs.

| MPC_ACC_HOR | main setpoint | main vehicle | exact setpoint | exact vehicle |
|---|---|---|---|---|
| 3 (default) | -0.06 m | 0.85 m | 0.27 m | 1.42 m |
| 5 | -0.05 m | 0.33 m | 0.38 m | 1.45 m |

So the stop got worse, and mission time didn't improve either since MPC_XY_CRUISE caps the speed long before the braking limit matters. On main the setpoint stops just short of the point, with the exact formula it runs past it because the planner has no margin left to react, and the vehicle trails that harder stop by about another metre. The old formula's extra room has been doing real work.

So I would keep this one small. Make the speed rise with the acceleration limit like it should, but never plan a shorter stop than today, and check in SITL that the overshoot stays where it is. The fully exact version would need a margin or a look ahead in the planner to be safe. If that sounds good to you, I can put up a PR with it.

Longer term the cleaner fix is probably planning straight to the target state each cycle, position and zero speed at the waypoint, the way Ruckig style generators do, instead of a speed limit from distance that the velocity smoothing chases. That's a much bigger change though, so just putting it out there.

dakejahl, 2026-10-05:
I'm not working on this. I'm keeping the issue alive because it should eventually be fixed.

It's not clear to me what the conclusion is. When actual acceleration is passed we overshoot because drag compensation is set incorrectly in SIH? 

> Collision prevention and the rovers don't have it, so I'd keep collision prevention on a worst case and assume zero for the rovers

Can it be added? I imagine collision prevention would benefit a lot from this correction.

If you figure it out please open a PR. Or open a draft PR if you want some review/collaboration.