PX4/PX4-Autopilot #24776: [Bug] PX4 v1.15.0 time syc issue observed in simulation with gz-garden.

ashirsat, 2025-04-29:
### Describe the bug

Hi all,
Me and my teammates were running gz-garden (ros2 humble) sim with PX4 v1.15.0 to test our navigation/planning code that switches the PX4 into Offboard mode to send setpoints on the /fmu/in/trajectory_setpoints topic.
The mavlink history at the bottom of the px4 log cycles between "time sync no longer converged" to "timesync converged".
The observed anomalous behavior is that the quadcopter keeps 'drifting'. The state-estimate can be seen swaying away (Ignore the jumps in the setpoint plot, the setpoint keeps snapping to the estimate because of a function in our planner when it kind of detects that the state-estimate is lagging)
This problem does not occur when we use PX4 1.14.3
Could anyone please help us with the solution to the time-sync issue and the weird drifting of the state-estimate?
For reference:
PX4 Log: https://review.px4.io/plot_app?log=c0559f4b-5092-42e7-b0aa-257c9e96293b
screenshot of the mavlink logger has been attached
GPS has been enabled in sim
Thanks a lot

### Screenshot / Media

[image]

### Flight Log

https://review.px4.io/plot_app?log=c0559f4b-5092-42e7-b0aa-257c9e96293b

### Software Version

PX4 v1.15.0
GZ-garden
Ubuntu 22.04
ROS2 Humble

### Vehicle type

None

### Additional context

Here is the link to the discord thread for the same issue.

-  https://discord.com/channels/1022170275984457759/1029122078948999238/1366591803457011864

beniaminopozzan, 2025-04-29:
Hi @ashirsat , when using GZ and ROS 2 together plese consider giving a look at https://docs.px4.io/main/en/ros2/user_guide.html#ros-gazebo-and-px4-time-synchronization

During GZ simulation PX4 takes its clock from the simulator itself, this allows to run the simulation at arbitrary real_time_factor values.  By defaul a target real time factor equal to one is set but it is harly achieved, rtf will likely be 0.98 - 0.99.
This means that PX4 clock is slightly slower than the "wall clock".
At the same time the uxrcedds_client sync (`UXRCE_DDS_SYNCT`) PX4 clock with ~~ROS 2~~ agent system clock (the wall clock).
Because the two run at different speed the offset will gradually increase until the sychronizers detects the jump and resets.

The best solution is to turn off the sychronizer and instruct ROS to use GZ clock. Regardless PX4 or not-PX4 application, when running ROS + Gazebo you should always run ROS under the control of GZ clock

TL;DR

- `[UXRCE_DDS_SYNCT](https://docs.px4.io/main/en/advanced_config/parameter_reference.html#UXRCE_DDS_SYNCT)` set to `false`
- use `gz_bridge` to bridge the GZ clock over to ROS 2 and turn on `use_sim_time` on each node.

ashirsat, 2025-04-30:
Hi @beniaminopozzan,

Thanks for the quick response. I did already try the solution you mentioned. I have disabled `UXRCE_DDS_SYNCT` another flight test as well with the other px4 params remaining the same. Also ran `gz_bridge` to bridgfe the gz clock to ROS. I was wondering how to add `use_sim_time` to ros2 nodes.

beniaminopozzan, 2025-06-02:
`use_sim_time` is just a ROS 2 parameter, so you set is as any other ROS 2 parameter.

If you disabled the synchronizer then you should not see those info and warning messages