PX4/PX4-Autopilot #24830: [Bug] Drone Crash Due to  Software Crash

liu12344321, 2025-05-12:
### Describe the bug

The drone suddenly fell from the sky as if it had completely lost power. After reviewing the flight logs, I found some software errors, and the RAM usage appeared abnormal.
Blow is hard fault detail:

[image]

### Flight Log

https://review.px4.io/plot_app?log=493454d6-f936-4068-b3cd-53738a62e02e

### Flight controller

HOLYBRO 6x

### Vehicle type

Hybrid VTOL

afwilkin, 2025-05-12:
Just flying in position flight mode with quadcopter? were you trying to transition? Or was this just hovering

liu12344321, 2025-05-13:
> Just flying in position flight mode with quadcopter? were you trying to transition? Or was this just hovering

I first took off in position mode using the multicopter mode, and everything seemed normal. Then I attempted the transition to fixed-wing mode, and the aircraft appeared to lose power at the exact moment I clicked the switch. However, when I checked the log, I didn’t see any record of the aircraft actually attempting to switch modes. As shown in the figure, the VTOL section is entirely blue, and there is no recorded output on motor5 (the tail motor).

[image]

liu12344321, 2025-05-13:
Before this crash, I had also performed several transitions with this aircraft—from multicopter to fixed-wing and back—and everything worked normally. It was only this time that something went wrong. I’m not sure what caused it, and I haven’t been able to reproduce the issue since, which makes it quite tricky to troubleshoot.

afwilkin, 2025-05-13:
I have seen that hard fault before as well. Have you checked out the tutorials on debugging hard faults in PX4? 

I would assist but a bit busy atm

closed 2026-01-12 by github-actions[bot]